microsoft/AirSim #1507: How to inherit ACarPawn?Met a build problem

mingqianliu, 2018-10-31:
i want to **inherit ACarPawn**，but i cant find this class in unreal，so i added AIRSIM_API in the declaration of ACarPawn class like this:
[image]
now i can create my own car pawn class inherit this class,like this:
[image]
but i still cannot build successfully.
Error message is claimed like this:
[image]

So how to create a new C++ class inherit ACarPawn in unreal？

trying to make clear expression,but English is not mother language.
**any suggestion or idea is appreciated,thanks a lot!**

mingqianliu, 2018-11-08:
it's ok to build a class inherit  carPawn inside the "plugin" folder,usually we don't want to make changes inside the plugin,but ..barely this method is acceptable

still looking forward to know other solutions

jonyMarino, 2020-08-12:
@ghost were you able to solve your issue?