microsoft/AirSim #1510: GPS fix jumping with MAVROS HITL

mikeisted, 2018-11-01:
Hi all,

The GPS fix reported by MAVROS in my HITL setup up jumps randomly. 

I am investigating the use of UE4/AirSim to test a multi-drone autonomous system.  FCU is Pixhawk cube, running PX4 stack and with Jetson TX2 companion computer.  TX2 is running ROS, connected using a MAVROS node over UART.

OK, so everything is set up (using the free mountains landscape atm) and manual control via transmitter looks good.  Autonomous flight control seems to work, but the GPS fix jumps randomly, of course throwing failsafes and making the simulator impossible to use.

Hooking up with QGroundcontrol seems to confirm that local body frame data is good (pose etc). but that the GPS seems to reset to the original (hardcoded) location.  However, in simulated flight, visual action remains smooth and it would not be clear this is happening at all.

RVBldr, 2018-11-21:
Internal VSTF 7696

yashkaushal, 2018-12-07:
Hi mike,

I faced the same issue and after looking around i found that this issue was not there in AirSim version 1.1 and it is present in version 1.2.0 and also in the pre-release of AirSim version 1.2.1.

I found a fix for this which works for version 1.2.0(not tested on v1.2.1), you may refer it at issue #1575 

Hope this helps,
Regards,
Yash Kaushal

madratman, 2018-12-19:
hey @mikeisted can you try pulling in the latest airsim code and let us know if this was resolved or not?

madratman, 2019-04-04:
closing as [this was fixed](https://github.com/Microsoft/AirSim/issues/1575#issuecomment-448123573)

closed 2019-04-04 by madratman