microsoft/AirSim #1511: questions about coordinate system of car and sensor in Airsim

dahewujiang, 2018-11-02:
hi, 
I am confused with the coordinate system in Airsim. The questions are as following:
1. where is the origin point of  the coordinate system in the car adopted by airsim?
2. which coordinate system is used by lidar points cloud?
3.when i use the "record" button to save data, it can also feedback the car's position. which coordinate system is used by this position? 

I think that lidar points cloud and position returned by "record" button are useing different coordinate systems because the "z" axis value of lidar points is negative while the value returned by button is positive with respect to  the same point.

bmalhigh, 2018-11-02:
@dahewujiang -- for the Lidar point-cloud, I am looking at changing it as part of issue https://github.com/Microsoft/AirSim/issues/1498 

Currently, the lidar point-cloud is in vehicle interial frame in NED coordinates. The vehicle interial frame has its origin at where the vehicle is spawned. I am trying to understand what frame it is best to use for lidar data and be consistent with other data returned by AirSim.

dahewujiang, 2018-11-05:
Hi，@bmalhigh 
I have seen the issue[#1498](https://github.com/Microsoft/AirSim/issues/1498).But I am still confused. 
1. when i set the play start position [x,y,z] to [0,0,0] and the road high to zero in UE4 world coordinate system and set the car position [x,y,z] to [0,0,-1] in settings.json, I get the car position [x,y,z] is [-0.005040, 0.000003,0.200948] by the "record" button. I donnot know the returned position is which point in the car. is it the mass point of the car? or the origin point of coordinate system of the vehicle car? is the value "0.200948" means that the direction is down?

2. how do i calculate the longitudinal and lateral distance between the cars when  running simulation? You know that  it's necessary to know those parameters in some scenarios when  testing.

Best regards!

RVBldr, 2018-11-21:
Internal VSTF 7693

closed 2020-12-18 by jonyMarino