microsoft/AirSim #1516: Simulator crashes after arming (PX4 SITL)

petergerten, 2018-11-04:
AirSim: master
PX4: master
OS: Ubuntu 16.04

I set up my AirSim project with PX4 SITL and I succesfully get MAVLink data in QGroundControl.
As soon as I issue the arm command (and I still get and "armed" back in QGC) - AirSim crashes.

Most of the time I get:
```
terminating with uncaught exception of type std::runtime_error: UdpClientPort socket send failed with error: 111

Signal 6 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=196655 
Malloc Size=41839 LargeMemoryPoolOffset=238511 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
```

Sometimes I get:
```
terminating with uncaught exception of type std::runtime_error: Message length 32 doesn't match expected length33

Signal 6 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=6
Malloc Size=65535 LargeMemoryPoolOffset=196655 
Malloc Size=41624 LargeMemoryPoolOffset=238296 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
```

But it reproducibly crashes every single time. Before I switched to the PX4 SITL I used the simple flight controller setup which runs stable with my current setup. If I don't arm the system but just let the simulation run and produce mavlink output - everything works and does not crash.

petergerten, 2018-11-04:
This is probably a very dirty hack but I just tried what would happen if I comment out
```
		if (hr == SOCKET_ERROR)
		{
			hr = WSAGetLastError();
			// perhaps the client is gone, and may want to come back on a different port, in which case let's reset our remote port to allow that.
			remoteaddr.sin_port = 0;
			throw std::runtime_error(Utils::stringf("UdpClientPort socket send failed with error: %d\n", hr));
}
```

from UdpClientPort.cpp 

This seems to work for me. Of course it does not solve the underlying issue but it does not crash the application anymore.

petergerten, 2018-11-23:
@RVBldr can you clarify what that means?

RVBldr, 2018-11-23:
We're tracking this issue on our internal Visual Studio bug database.

petergerten, 2020-01-21:
@clovett just pulled airsim master (without my quick&dirty patch mentioned above) and noticed that this issue still persists.

lovettchris, 2021-06-04:
This PR might also fix the issue:
https://github.com/microsoft/AirSim/pull/3757

closed 2021-06-18 by zimmy87

zardosht, 2023-10-26:
Setting the parameter `MAV_PROTO_VER` to "Always use version 2.0" in PX4 fixes the problem with message length runtime error.