microsoft/AirSim #1517: Recommended PX4 firmware version for SITL

petergerten, 2018-11-04:
I tried master (1.8.X) and get:

`ISSUE: EKF baro hgt timeout - reset to baro`

I also found in https://github.com/Microsoft/AirSim/blob/master/docs/px4_build.md 'we are currently using v1.6.0rc1' - and tried it:

I don't get the above error anymore but instead get:

```
ERROR [commander] CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBRATION
WARN  [commander_tests] Failsafe enabled: no global position
```

Same issue as mentioned here: https://github.com/PX4/Firmware/issues/10517

Is there any PX4 firmware version recommended to work with the current AirSim?

petergerten, 2018-11-04:
it turns out there is a workaround for the "CRITICAL NAVIGATION FAILURE" here: https://github.com/Microsoft/AirSim/issues/1396
But I would still be interested what other would recommend for a stable PX4 SITL setup.

timesofbadri, 2018-11-19:
1.5.5 ([https://github.com/PX4/Firmware/releases/tag/v1.5.5]) works for me in HITL. Try this.

lovettchris, 2018-12-18:
I have test v1.8.2 works fine in SITL mode.

james-mchugh, 2019-01-10:
On v1.8.2, I am also getting 
`ISSUE: EKF baro hgt timeout - reset to baro`

On v1.5.5 I get
```
ERROR [commander] CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBRATION
WARN  [commander_tests] Failsafe enabled: no global position
```

jonyMarino, 2020-08-14:
Closed due to age

closed 2020-08-14 by jonyMarino