microsoft/AirSim #1521: Can't connect to drone with Python on new install

abe-mart, 2018-11-07:
I'm having trouble with a fresh download of AirSim in a new conda environment, using the latest binaries.  After starting the unreal binary, I run hello_drone.py, but it just hangs forever at the client.confirmConnection() command.  This has always worked fine before, so I'm not sure what has changed.  Forcing an exit gives the following traceback:

```
Traceback (most recent call last):
  File "C:\Users\Civil\Downloads\Airsim2\AirSim-master\PythonClient\multirotor\hello_drone.py", line 11, in <module>
    client.confirmConnection()
  File "C:\Users\Civil\Downloads\Airsim2\AirSim-master\PythonClient\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "C:\Users\Civil\Downloads\Airsim2\AirSim-master\PythonClient\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "C:\Users\Civil\AppData\Local\conda\conda\envs\airsim_test\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\Civil\AppData\Local\conda\conda\envs\airsim_test\lib\site-packages\msgpackrpc\future.py", line 25, in get
    self.join()
  File "C:\Users\Civil\AppData\Local\conda\conda\envs\airsim_test\lib\site-packages\msgpackrpc\future.py", line 22, in join
    self._loop.start()
  File "C:\Users\Civil\AppData\Local\conda\conda\envs\airsim_test\lib\site-packages\msgpackrpc\loop.py", line 22, in start
    self._ioloop.start()
  File "C:\Users\Civil\AppData\Local\conda\conda\envs\airsim_test\lib\site-packages\tornado\ioloop.py", line 863, in start
    event_pairs = self._impl.poll(poll_timeout)
  File "C:\Users\Civil\AppData\Local\conda\conda\envs\airsim_test\lib\site-packages\tornado\platform\select.py", line 63, in poll
    self.read_fds, self.write_fds, self.error_fds, timeout)
KeyboardInterrupt
```

Is there some trouble with the tornado package perhaps?  I've tested this with LandscapeMountains and ZhangJiaJie.

dahewujiang, 2018-11-08:
Have you placed the "plugins" in your UE4 project?

abe-mart, 2018-11-08:
In this case I'm using the pre-built binaries released on this site.

abe-mart, 2018-11-09:
I'm still not sure what the problem was, but I was able to resolve this issue by installing from setup.py instead of the pip version.

closed 2018-11-09 by abe-mart