microsoft/AirSim #1522: v1.2.1 Release not working

damien022, 2018-11-08:
Tried running the latest release commit on Windows, tried Plain and TrapCam, with no settings.json
When selecting the multirotor, I get the error message
[image]

`Error at startup: Failed to load asset class- Class /AirSim/Blueprints/BP_FlyingPawn.BP_FlyingPawn_C`

damien022, 2018-11-08:
Hi DoC! thank you for the quick reply. I do not have access to the Editor as I am running the built binary from the v1.2.1 pre-release

saisubramaniam147, 2018-11-10:
Possible to upload the fixed binary soon? Thanks.

RVBldr, 2018-12-06:
Please try and re-download the binary, this should be resolved.