microsoft/AirSim #1523: is there a plan to develop ultrasonic sensor and millimeter-wave radar?

dahewujiang, 2018-11-09:
hi, i think that the types of sensor model in airsim is too less to develop algorithms for auto driving. I think the ultrasonic sensor and millimeter-wave radar is necessary.But it too complex for me to model sensor. is there any plan to develop those sensors ?

RVBldr, 2018-12-06:
We'll add that to our internal feature list.

closed 2018-12-18 by dahewujiang

flynneva, 2020-07-21:
was this completed already?

JuanshuB2, 2022-03-09:
Hi! I think it is interesting to mention this project to simulate a FMCW radar in Unity. It generates the raw data that can be further processed:  https://github.com/chstetco/virtualradar. However, it is not available in Unreal Engine, so I am wondering how challenging can it be.