microsoft/AirSim #1525: DQNcar.py MemoryError(Ubuntu18.04)

yujianyuanhaha, 2018-11-10:
OS: __Ubuntu18.4__
CNTK version __2.5__
UE4Editor version: 4.17
Airsim version: 4.18
I finally successfully install the cntk on Ubuntu18.04, and `hello_car.py` also runs well, while met new issue of __MemoryError__ when call the `DQNcar.py`, error shown as
```
/home/jet/anaconda3/envs/py2/lib/python2.7/site-packages/cntk/cntk_py_init.py:47: UserWarning: Unsupported Linux distribution (ubuntu-18.04). CNTK supports Ubuntu 16.04 and above, only.
  warnings.warn('Unsupported Linux distribution (%s-%s). CNTK supports Ubuntu 16.04 and above, only.' % (__my_distro__, __my_distro_ver__))
Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Traceback (most recent call last):
  File "DQNcar.py", line 505, in <module>
    agent = DeepQAgent((NumBufferFrames, SizeRows, SizeCols), NumActions, monitor=True)
  File "DQNcar.py", line 270, in __init__
    self._memory = ReplayMemory(memory_size, input_shape[1:], 4)
  File "DQNcar.py", line 35, in __init__
    self._states = np.zeros((size,) + sample_shape, dtype=np.float32)
MemoryError

```
Anybody got a idea?

ZhanPython, 2019-05-09:
Hi, I meet with the same issue, have you figured it out?

yujianyuanhaha, 2019-05-10:
> Hi, I meet with the same issue, have you figured it out?

nope. I later refer to some other open source code in github, tho older library

subinlab, 2019-05-12:
I have same issue also in `DQNdrone.py`

rajat2004, 2019-10-10:
Is this related to https://github.com/microsoft/AirSim/issues/2226?

MoritzGa, 2019-10-11:
Well how big do you choose your replay buffer to be? I mean I can't fit 1 million  samples in there as they did in the paper.

rajat2004, 2019-10-11:
I can't say that either :-)
Seems to be trial and error (if this is causing the error)

zimmy87, 2021-01-07:
possibly related to #1009

zimmy87, 2021-01-15:
Hi, thank you for reporting this issue.  This script was recently removed in #3215 and replaced with a new script at \AirSim\PythonClient\reinforcement_learning\dqn_car.py that uses the new airgym module. I tested the new script locally and it runs for me. Can you retest with the new script and report back if you see any issues?

closed 2021-01-15 by zimmy87