microsoft/AirSim #1528: can i just download the CityEnviron.zip and use it in linux?

1453042287, 2018-11-12:
just like the title, thanks a lot!

tzimasak, 2018-11-13:
Sadly no. The environments in the releases section are to be used in Microsoft Windows.

kang235, 2018-11-13:
Will there be linux version released?

sytelus, 2018-11-16:
Try this out: https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux

1453042287, 2018-11-18:
@sytelus hi there, when i use the 7zip to extract the CityEnviron.zip.001, it report errors:
7za x CityEnviron.zip.001

7-Zip (a) [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=zh_CN.UTF-8,Utf16=on,HugeFiles=on,64 bits,12 CPUs Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz (906EA),ASM,AES-NI)

Scanning the drive for archives:
1 file, 1782579200 bytes (1700 MiB)

Extracting archive: CityEnviron.zip.001
           
CityEnviron.zip
ERRORS:
Headers Error
Unconfirmed start of archive

CityEnviron.zip
WARNINGS:
There are data after the end of archive

--
Path = CityEnviron.zip.001
Type = Split
Physical Size = 1782579200
Volumes = 2
Total Physical Size = 3117962657
----
Path = CityEnviron.zip
Size = 3117962657
--
Path = CityEnviron.zip
Type = zip
ERRORS:
Headers Error
Unconfirmed start of archive
WARNINGS:
There are data after the end of archive
Physical Size = 3117932051
Tail Size = 30606

ERROR: Data Error : CityEnviron/Content/Paks/CityEnviron-LinuxNoEditor.pak
                                                               
Sub items Errors: 1

Archives with Errors: 1

Warnings: 1

Open Errors: 1

Sub items Errors: 1

can you give me some suggestions? thanks a lot!

kang235, 2018-11-19:
> Try this out: https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux

Thanks a lot, @sytelus .

closed 2018-12-06 by madratman