microsoft/AirSim #1538: Multiple Quadrotors control using API acts as a group of 4 instead of all together

yashkaushal, 2018-11-14:
Hi,
I tried to make a swarm of drones take off and land together.

I configured settings.json to have 12 drones in Unreal and used python API to make them takeoff together, but instead of all taking off together random 4 take off first, then next 4 and so forth.

See screenshot for reference:
[image]
All were expected to take off together but only the first 4 took off, after a few seconds the next 4 took off.

My OS is windows 10 pro version 1803, AirSim version 1.2(today's build), python 2.7.15, unreal version 4.20

Below is the settings.json I used:
``` json
{
    "SimMode": "Multirotor",
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SettingsVersion": 1.2,
    "Vehicles": {
        "SimpleFlight1": {
            "VehicleType": "SimpleFlight",
            "X": 4,
            "Y": -15,
            "Z": -2,
            "DefaultVehicleState": "Disarmed",
            "EnableCollisionPassthrogh": false,
            "EnableCollisions": true,
            "AllowAPIAlways": true,
            "RC": {
                "RemoteControlID": 0,
                "AllowAPIWhenDisconnected": false
            }
        },
        "SimpleFlight2": {
            "VehicleType": "SimpleFlight",
            "X": -4,
            "Y": -15,
            "Z": -2,
            "DefaultVehicleState": "Disarmed",
            "EnableCollisionPassthrogh": false,
            "EnableCollisions": true,
            "AllowAPIAlways": true,
            "RC": {
                "RemoteControlID": 0,
                "AllowAPIWhenDisconnected": false
            }
        },
        "SimpleFlight3": {
            "VehicleType": "SimpleFlight",
            "X": 0,
            "Y": -15,
            "Z": -2,
            "DefaultVehicleState": "Disarmed",
            "EnableCollisionPassthrogh": false,
            "EnableCollisions": true,
            "AllowAPIAlways": true,
            "RC": {
                "RemoteControlID": 0,
                "AllowAPIWhenDisconnected": false
            }
        },
        "SimpleFlight4": {
            "VehicleType": "SimpleFlight",
            "X": 4,
            "Y": -10,
            "Z": -2,
            "DefaultVehicleState": "Disarmed",
            "EnableCollisionPassthrogh": false,
            "EnableCollisions": true,
            "AllowAPIAlways": true,
            "RC": {
[... 284 lines cut ...]
# that's enough fun for now. let's quit cleanly
client.enableApiControl(False, "SimpleFlight1")
client.enableApiControl(False, "SimpleFlight2")
client.enableApiControl(False, "SimpleFlight3")

client.enableApiControl(False, "SimpleFlight4")
client.enableApiControl(False, "SimpleFlight5")
client.enableApiControl(False, "SimpleFlight6")

client.enableApiControl(False, "SimpleFlight7")
client.enableApiControl(False, "SimpleFlight8")
client.enableApiControl(False, "SimpleFlight9")

client.enableApiControl(False, "SimpleFlight10")
client.enableApiControl(False, "SimpleFlight11")
client.enableApiControl(False, "SimpleFlight12")

```

I tried using single client for each quad instead of a single client being shared among them but that also yielded the same result.

Please suggest me what steps to take to make them take off together.

Thanks,
Yash Kaushal

RVBldr, 2018-11-15:
Not that this is the actual cause, but did you catch that you do a typo in the first set of client.disarm:  
client.armDisarm(True, "SimpleFlight10")
client.armDisarm(True, "SimpleFlight11")
client.armDisarm(True, "SimpleFlight10")

yashkaushal, 2018-11-18:
Hi @RVBldr 
Thanks for pointing the typo,
I did some searching around and found that it has been reported earlier as well in the last comment of #1137 by KavehFathian.

RVBldr, 2018-11-18:
in the video referenced in #1137, it's interesting that they are able to work 9 apparently without the 4 ship bug. Internally we're taking at look at that and initial thoughts are that it's a threading issue.

sytelus, 2018-11-21:
I think the issue is in [this line](https://github.com/Microsoft/AirSim/blob/master/AirLib/src/api/RpcLibServerBase.cpp#L260).

`pimpl_->server.async_run(4);   //4 threads`

Try changing it to number of vehicles and see if it works. 

We need to set the number of threads from AirSimSettings. Ideal would to do max of count of vehicles in AirSimSettings or user specified value.