microsoft/AirSim #1539: API control not working in AirSim Unity

yashkaushal, 2018-11-14:
Hi,

I want to control Quadrotor in Unity using API commands and tested various API commands with it.
The moveByRC command is not working in AirSim Unity it does not give any error as well.

## Steps to reproduce:
- Build AirSim for Unity and launch drone demo
- Run `manual_mode_demo.py` 
- Drone will take off but will not change yaw, pitch or roll

I tested it with other commands as well and the commands which use any AirSim struct - yawmode, drivetrain or rcdata are not working from API calls. (I've not tested yawmode and drivetrain commands with the latest build).

I'm using Windows 10 pro version 1803, AirSim version 1.2(today's build), python 2.7.15, unity version 2018.2.8.

Thanks,
Yash Kaushal

josedeo, 2018-11-19:
Hi Yash,
Would you mind making sure the setttings.json file under %USERPROFILE%\Documents\AirSim look something like this below? It seems to be working for me with these settings.
If its not like that, can you give it a try?
Thanks
--- Jose

Here's what I have in mine:
{
  "SimMode": "Multirotor",
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      }
    }
   }
}

RVBldr, 2018-12-06:
@yashkaushal Can you confirm that this is resolved?

yashkaushal, 2019-01-03:
I'm no longer using Unity with AirSim but testing using Unreal with AirSim build from source of 14-Dec-2018 is giving a very jerky (almost vibrating)movement of QuadRotor.

ho0-kim, 2019-01-28:
I have the same issue.
With moveByRC, I can't control pitch, roll and yaw.
It works only with throttle level and it vibrates when it moves up as @yashkaushal mentioned.