microsoft/AirSim #1540: Multiple drones in AirSim Unity?

yashkaushal, 2018-11-14:
Hi,

Does Unity AirSim beta support multiple drones? 
I tried settings.json for multiple drones and got error window saying `Problem in starting AirSim server - please check logs for more information`

I checked the logs and this looked like the reason: `Error at startup:  VehicleSetting for vehicle name SimpleFlight was requested but not found Vehicle=SimpleFlight`

Below is the complete log file.
[Editor log.zip](https://github.com/Microsoft/AirSim/files/2580751/Editor.log.zip)

Thanks,
Yash Kaushal

josedeo, 2018-11-20:
Hi Yash
Thanks for raising this point! 
AirSim on Unity doesn't support multiple drones in its current implementation. 
Please consider contributing on this feature! We'd love some community help!
Thanks!
--- Jose

MelSoS, 2019-01-08:
Is this still an issue, or is there already a solution to the multi-drone problem?
Thank you and BR

sousheel, 2019-01-29:
Hi @MelSoS and @josedeo , I am working on the problem right now. Let me know if you have any pointers. My fork is at https://github.com/cohrint/AirSim.

MelSoS, 2019-01-30:
@sousheel nice to here that - we are getting more!
Unfortunately, there is no update yet (as far as I now), let's inform each other in this regard. 
Best,

jasonhbartlett, 2019-01-31:
I suspect this is partly related to this issue I submitted #1696 where I believe the structs located in AirSim/Unity/UnityDemo/Assets/AirSimAssets/Scripts/Utilities/AirSimSettings.cs script need to be updated to be able to read the "Vehicles" data structure.

jonyMarino, 2022-05-30:
Fixed with #3128

closed 2022-05-30 by jonyMarino