microsoft/AirSim #1541: Build Error MavLinkCom C1001

Waarbubble, 2018-11-14:
Having just re Installed my windows 10 PC I proceeded to install 

- Visual Studio 2017 enterprise v. 15.9.0, with windows 8.1 SDK
- Unreal Engine 4.21.0
 - AirSim

but running the build.cmd I get the following error:

> r:\program files\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error has occurred i n the compiler. [R:\Program Files\AirSim\MavLinkCom\MavLinkCom.vcxproj] 
  (compiler file 'd:\agent\_work\3\s\src\vctools\compiler\cxxfe\sl\p1\c\types.c', line 4563)
   To work around this problem, try simplifying or changing the program near the locations listed above.
  Please choose the Technical Support command on the Visual C++
   Help menu, or open the Technical Support help file for more information
  r:\program files\airsim\mavlinkcom\common_utils\type_utils.hpp(59): note: see reference to class template instantiati
  on 'mavlink_utils::type_utils::detail::has_begin_end<T>' being compiled
  ThreadUtils.cpp
  AdHocConnection.cpp
  AdHocConnectionImpl.cpp

Any one know what I am d

micmarty, 2018-11-14:
I have the exact same issue. It looks like we're missing some Visual Studio component or something... I'll let you know when I find the solution :)

forestscorpion, 2018-11-15:
i have the exact same issue too -.-

simon-achatz, 2018-11-15:
I'm having the same issue, already when running 'build.cmd' in root folder the first time.

arguskos, 2018-11-15:
Same error for me, but even from the root folder of the AirSim

nicksunyang, 2018-11-15:
I reinstall vs2017 , the problem can't be fixed

nicksunyang, 2018-11-15:
i also use Unreal Engine 4.18.3, uninstalled  4.19.2

Waarbubble, 2018-11-15:
A work around i found was to install visual studio 2017 v 15.6.8

madratman, 2018-11-15:
Hey everyone. Yeah, the issue is with VS 2017 15.9.0, which was released a couple days back on Nov 13. 
I tested with 15.8.9, and things do compile. Please uninstall 15.9.0 and install 15.8.9, as explained here (links for 15.8.9 therein)
https://docs.microsoft.com/en-us/visualstudio/productinfo/installing-an-earlier-release-of-vs2017

We're looking into what is causing the error at Microsoft, please use 15.8.9 till then. Thanks @Waarbubble for reporting the issue and the fix.

jindegithub, 2018-11-18:
I have the same errors ,and my Installation Environment is vs2017_15.3 +Unreal version 4.18.do  I change vs2017_15.3  to vs2017_15.8.9?

jindegithub, 2018-11-20:
> @jindegithub Did you install win sdk 8.1 and VC++, while installing VS 2017?

Thank you for your reply! I added C++ development desktop, Windows 8.1 SDK and Unreal Engine Installer while installing vs2017 professional.

madratman, 2018-11-20:
@jindegithub hmm, that's weird that it still is not compiling. 
@bmalhigh fixed the issue causing the C1001 error as can be seen [here](https://github.com/Microsoft/AirSim/pull/1558). Can you try a `git pull --rebase` and `build.cmd` again?

james-mchugh, 2018-11-21:
@madratman I have been able to get it to build now, but like @jindegithub, I too had to install Unreal Engine Launcher through VS despite having it installed through Epic Games Launcher. It looks like this issue has been experienced by someone else as well #1188. I am not sure why this is.

closed 2019-02-19 by madratman

AraHaan, 2019-07-17:
Hmm I actually wonter if the C++ compiler it open source: I ran into this issue myself when I:
1. created an empty static lib project.
2. enabled modules TS and added ``/module:export`` as an additional option.
3. Enabled PDB with edit and continue and the options related to that.

Results:

```
1>d:\agent\_work\4\s\src\vctools\Compiler\CxxFE\sl\p1\c\ModuleWriter.cpp:4862: sorry: not yet implemented
1>src\_internaltypes.cxx(788): fatal error C1001: An internal error has occurred in the compiler.
1>(compiler file 'd:\agent\_work\4\s\src\vctools\Compiler\CxxFE\sl\p1\c\ModuleUtilities.h', line 25)
1> To work around this problem, try simplifying or changing the program near the locations listed above.
1>Please choose the Technical Support command on the Visual C++
1> Help menu, or open the Technical Support help file for more information
```