microsoft/AirSim #1544: AirSim Unity - Keyboard controls not working

dza6549, 2018-11-15:
Using Unity 2018.2.15f1  on Win 10 Home
Tried AirSim plugin build with latest (today) Win 10 SDK 10.0.17763.1 - build failed.
Changed target to 10.0.15063.1 - build succeeded with warning.
DroneDemo scene did not work directly - tried SimModeSelector - scene loaded.
Error in Editor - Cannot read settings file  C:\Program Files\Unity\Editor\settings.json Vehicle=SimpleFlight
UnityEngine.Debug:LogError(Object)

Scene runs but drone does not respond to keyboard commands.
Car does not respond to keyboard commands.

Let me know if I can provide further info.

Thanks!

TreyYi, 2018-11-15:
Hi dza6549,

The sample scenes are not even run when I clicked "Play" Button, and it crashes Unity.. Any idea?

madratman, 2018-11-15:
@dza6549 you need Win SDK **8.x**, as mentioned [here](https://microsoft.github.io/AirSim/docs/build_windows/) and [here](https://github.com/Microsoft/AirSim/tree/master/Unity#build-airsim)

dza6549, 2018-11-16:
@madratman thanks for your reply. I have tried 8.1 and I get the following fail build:

"..Path\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (Clean;Build target) (1) ->
"..Path\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (default target) (2:2) ->
"..Path\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj" (default target) (5:2) ->
(ClCompile target) ->
  c:\program files (x86)\windows kits\8.1\include\um\combaseapi.h(229): error C2760: syntax error: unexpected token 'identifier', expected 'type specifier' [..Path\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj]

    0 Warning(s)
    1 Error(s)

dza6549, 2018-11-16:
Closing this one.
More details over here [https://github.com/Microsoft/AirSim/issues/1550](https://github.com/Microsoft/AirSim/issues/1550)

closed 2018-11-16 by dza6549