microsoft/AirSim #1545: error C2220: warning treated as error - no 'object' file generated

TreyYi, 2018-11-15:
Hi,

The error was issued in the past post ([here ](https://github.com/Microsoft/AirSim/issues/76)), but I found that encoding error is still caused in some regions.

I've tried to compile the AirSim for Unity integration, but noticed there's small error when compiling.
If you set your region to "Korea", not "United State", then you would get this error message on Developer Command Prompt for VS 2017:
"error C2220: warning treated as error - no 'object' file generated"

However,
1. as you change your region to "United State", you can simply fix the error.
2. Or you can follow some suggestion in [here](https://github.com/Microsoft/AirSim/issues/76)

Thanks,
Trey

allusernamestakenexceptthis, 2018-11-16:
Thanks, I have the same problem.  just replacing the quotes in “AS IS” with normal quotes work.  weird they'd use such thing in a license.

hyhou, 2018-11-16:
Change region didn't work in my case but replacing quotes worked. Thank you all!

closed 2018-12-14 by TreyYi