microsoft/AirSim #1546: [Crash] AirSim Unity - Unity 2018.2.15f1

TreyYi, 2018-11-15:
Hi,

I've tried to run the sample scenes that MS provides, but it crashes Unity.
I loaded the scenes such as Car, Drone, but as soon as I click the Play button, it crashes Unity.
[image]

Any suggestion?

Regards,
Trey

dza6549, 2018-11-15:
Hey Trey

my Unity crashed when using Win 10 SDK 10.0.17763.1.
10.0.15063.0 loaded but did not work :(  You saw my post.

You can check your target:
1: Open ..\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln with VS
2: In VS open Project .. Properties
3: What do you have?

[image]

TreyYi, 2018-11-15:
> AirsimWrapper

I have these options for Windows SDK version
[image]

However, when I typed the command below on PowerShell to check what version I have,
$(Get-Item "hklm:\SOFTWARE\Microsoft\Microsoft SDKs\Windows").GetValue("CurrentVersion")
[image]
My current version is 8.0.50727.

Please let me know if I can provide you with any additional information.

TreyYi, 2018-11-15:
Thanks for your prompt reply.
Though, It's still not working with different Windows SDK settings.
It just crashes Unity Application as soon as I click "Play" button..

dza6549, 2018-11-15:
Yeah. It is only 24 hours old. :)

Maybe we should wait until someone gets it working.

Cheers!

madratman, 2018-11-15:
@TreyYi  @dza6549  please use win sdk **8.1**, [Visual Studio 2017, 15.8.9](https://docs.microsoft.com/en-us/visualstudio/productinfo/installing-an-earlier-release-of-vs2017) and try again. 
sdk 8.1 was mentioned in the docs before. VS 2017 15.9.0 was released on Nov 13, and that is new. I have filed a PR an update to the doc. till than, you can look at the diff https://github.com/Microsoft/AirSim/pull/1549/files

dza6549, 2018-11-16:
@madratman Ah I see - I will update VS Community - I am using 15.65

Thanks

madratman, 2018-11-16:
@dazhang88 @TreyYi My bad. I think anything <=15.8.9 would work. I think your issue is not using windows sdk 8.1. Can you try with 15.65 + 8.1 and get back to us?

dza6549, 2018-11-16:
Just a heads up to anyone who is following this thread. The download links on the VS 2017 15.8.9 page mentioned above appear to be mixed up. Community.exe downloads vs_enterprise.exe and vice versa.

dza6549, 2018-11-16:
@madratman whoops missed your comment. And have uninstalled 15.6.5 already :)

So 15.8.9 it is :) Cheers

dza6549, 2018-11-16:
@TreyYi 

I got it sorta working over here [https://github.com/Microsoft/AirSim/issues/1550](https://github.com/Microsoft/AirSim/issues/1550)

See if this helps

Cheers!

dza6549, 2018-11-16:
@TreyYi Closing the thread is ok with me man. It is your thread! :)

TreyYi, 2018-11-16:
Yup you right. Thanks! Hope the Drone issue would be resolved soon. I'll close the thread.

closed 2018-11-16 by TreyYi

sab2018, 2018-12-07:
Windows 7 OS,
Unity 2018.2.14
VS 15.6.7

Just had a question on the SDK. I've tried all the solutions mentioned here. Was running VS 15.9. The build didn't succeed because I had PowerShell 3.1. Upgraded to 5.1. Then tried to rebuild with VS 15.6,7 + 8.1 SDK. had to get SDK 16299 as well after I got an error for it while building the unity project. Build succeeded but I'm getting the same screen as shown by @TreyYi  on unity and unity crashes as I press play. 
@dza6549 @madratman My target platform in VS is windows 10, with SDK version 16299. is that fine for a windows 7 OS?  maybe that's the reason my project crashes? I can't seem to find the cause for this.