microsoft/AirSim #1548: can you please release binary version under linux ?

18605973470, 2018-11-15:
as titled

sytelus, 2018-11-16:
Try this out: https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux

18605973470, 2018-11-16:
@sytelus 

I get this error : 
error while loading shared libraries: libopenvr_api.so: cannot open shared object file: No such file or directory.

I am new to UE4. So how can I solve this problem ?
Thanks a lot .

18605973470, 2018-11-16:
Hi @sytelus 
Do you have version 1.2.1 ? Maybe I need "no display" mode.

1453042287, 2018-11-19:
@18605973470 can you extract the CityEnviron.zip?

18605973470, 2018-11-19:
Hi @1453042287 

Yes. I use 7zip as suggested.

1453042287, 2018-11-24:
@18605973470 sorry to bother again, did you run the linux version successfully?

18605973470, 2018-11-24:
You should extract engine.zip and add corresponding path to LD path

1453042287, 2018-12-06:
@18605973470 em...i come back again, i download the CityEnviron.zip and extract it successfully, and i also add the path of .so in engine.zip to LD path, but when i run the CityEnviron.sh i still have the Segmentation fault (core dumped) error, is there something i missed?

jonyMarino, 2020-09-09:
treated on #1567

closed 2020-09-09 by jonyMarino