microsoft/AirSim #1550: Unity - Fix for Build Error C2760 in combaseapi.h - but still not working

dza6549, 2018-11-16:
Hi, still can't get the copter working. I did find a fix for the C2670 error though. Any thoughts appreciated.

Win 10 Home
Unity 2018.2.15f1
VS Community 2017 15.8.9
SDK 8.1

Unity/build.cmd Build fails with:
windows kits\8.1\include\um\combaseapi.h(229): error C2760: syntax error: unexpected token 'identifier', expected 'type specifier'

Fix suggested here:
[https://developercommunity.visualstudio.com/content/problem/185399/error-c2760-in-combaseapih-with-windows-sdk-81-and.html](https://developercommunity.visualstudio.com/content/problem/185399/error-c2760-in-combaseapih-with-windows-sdk-81-and.html)

1: Open (make backup first of course) C:\Program Files (x86)\Windows Kits\8.1\Include\um\combaseapi.h
2: Add struct IUnknown; // Workaround for "combaseapi.h(229): error C2187: syntax error: 'identifier' was unexpected here" when using /permissive- 
3: Put this line at top of file above includes
4: Rebuild 
5: Build succeeds but with warning:

Build succeeded.

"..Path\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (Clean;Build target) (1) ->
"..Path\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (default ta
rget) (2:2) ->
"..Path\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj" (default target) (5:
2) ->
(Link target) ->
  LINK : warning LNK4075: ignoring '/INCREMENTAL' due to '/LTCG' specification [..Path\AirSi
m\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj]

    1 Warning(s)
    0 Error(s)

6: Run project SimModeSelector Unity Project Warning:
Cannot read settings file  C:\Program Files\Unity\Editor\settings.json Vehicle=SimpleFlight
UnityEngine.Debug:LogError(Object)
AirSimUnity.Vehicle:PrintLogMessage(String, String, String, Int32) (at Assets/AirSimAssets/Scripts/Vehicles/Vehicle.cs:279)
AirSimUnity.VehicleCompanion:PrintLogMessage(String, String, String, Int32) (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:193)

7: Create settings.json file in Assets folder:
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor"
}

8: Run Unity Project SimModeSelector. Copter does not respond to Page Up/Down & WASD
Car responds to WASD

dza6549, 2018-11-16:
Update: settings.json must be placed in C:\Program Files\Unity\Editor folder

TreyYi, 2018-11-16:
Putting customized settings.json file in C:\Program Files\Unity\Editor folder would cause an issue like below:
[image]

So, I just deleted the settings.json file and at least the Car works.

madratman, 2018-11-19:
@dza6549 looks like you were able to resolve this? If so, let's close the issue.

msb336, 2018-12-14:
@dza6549 thanks for reporting your issue, as well as a fix! Latest PR should resolve the build break without having to modify any internal windows include files.

As for the drone flight issue, keyboard input is not yet supported on unity. Have you tested the AirSim APIs by running HelloCar or HelloDrone?

dza6549, 2018-12-14:
@msb336 I haven't tried the APIs yet. Will do so with the new build and let you know. Thanks

msb336, 2018-12-18:
@dza6549 
Have you had any problems related to this issue? If not, let's close it

dza6549, 2018-12-18:
@msb336 

OK let's close it. Thanks

closed 2018-12-18 by dza6549