microsoft/AirSim #1551: Setting up network for Airsim use

sebbom, 2018-11-16:
Dear developers,

I am wondering what is the correct setup to be able to have a windows machine on the network performing all the rendering while I am developing on an Ubuntu machine on which I use the Python or C++ API.
I specified the rendering machine IP when connecting to the client and make sure port 41451 is open both machine.
On the rendering machine I specified its IP in the settings.json and enabled RCP.

Unfortunately I am not able to connect between both computers.

All the best,

bmalhigh, 2018-11-16:
This really is more generally about your network setup (and/or corp policies of your environment). Not specific to AirSim.

e.g., in my environment I am able to run UE4 on my windows machine and run the python client on my Ubuntu machine. But can't do vice-versa because of domain enforced policies in my environment.

As you mentioned, when you run airsim client (using python or C++), you can configure the server IPAddress and port to connect to.

e.g., for python,
      client = arisim.CarClient(ip = "10.100.10.10", port = 41451)

sebbom, 2018-11-19:
Thanks for your feedback!
I was wondering if there is any change to be done on the json file in order to have the connection done.

I checked and the network setup should be fine.

closed 2018-12-06 by madratman

ivannson, 2020-02-19:
@sebbom Hi, I know this was a while ago, but did you manage to get this to work? I am trying to do the same thing, but unfortunately not much luck.