microsoft/AirSim #1552: How to run new environments like AirSimNH in linux. Please provide the instructions...

Vinay545, 2018-11-17:


damien022, 2018-11-19:
You need to modify the launch script to point to where the libraries are. You can find them in the Engine zip file.
In the bash script to launch the Environment, do
`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/libopenvr_api:/path/to/libopenal.so`

luqiang21, 2018-11-20:
@damien022 Hi thank you for the answer, but where can I find the libraries. I didn't find those two files on my computer.

Endf22, 2018-11-21:
Hello,

I had the same issue but finally, I have managed to open AirSimNH environment by following these steps:

1) Add Engine files to AirSimNH directory
2) Execute  AirSim.sh file in terminal (chmod +x AirSimNH.sh or sudo ./AirSimNH.sh)
2) Run AirSimNH  (AirSimNH/Binaries/Linux/AirSimNH)   

I hope it helps,

Vinay545, 2018-11-22:
@Endf22 Thank you. Its working now. 
@damien022  Thank you

ssokhey, 2019-01-07:
@Endf22 
Hi,

What exactly I have to do in Step 1. ?
I just can't figure out can you please explain.

madratman, 2019-01-15:
Hey, just updated AirSimNH and added a blocks binary. 
`Engine/`. `libopenar.so`, `libopenvr.so` should be all in there. 
Please check updated instructions at https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux

closed 2019-02-19 by madratman

alonyomtov123, 2019-11-01:
Where are the files that should be moved?
How are they called and to where?
Also, is there a way to use PythonClient in AirSimNH like in Airsim?
Thank You