microsoft/AirSim #1553: can not extract the CityEnviron.zip

1453042287, 2018-11-18:
like the title, when i use the 7zip to extract the CityEnviron.zip.001 for linux, it report errors:
Path = CityEnviron.zip
Type = zip
ERRORS:
Headers Error
Unconfirmed start of archive
WARNINGS:
There are data after the end of archive
Physical Size = 3117932051
Tail Size = 30606
ERROR: Data Error : CityEnviron/Content/Paks/CityEnviron-LinuxNoEditor.pak
Sub items Errors: 1
Archives with Errors: 1
Warnings: 1
Open Errors: 1
Sub items Errors: 1

luqiang21, 2018-11-20:
I encountered similar problem. When I extract it, it is requiring password.

1453042287, 2018-11-21:
i can't extract the file when i use a ubuntu environment but when i use the windows it works fine!

closed 2018-11-21 by 1453042287

nighthb2019, 2019-01-10:
Hey , I followed the operation, but still can't extract the CityEnviron.zip. I am running under WIN10. 
Does anyone know this reason?