microsoft/AirSim #1554: is there a way to get the instance segementation label?

1453042287, 2018-11-19:
like the title, i'm new here, thanks a lot!

madratman, 2018-11-19:
Please read [the segmentation section here](https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/docs/image_apis.md#segmentation) and [the section on getting mesh names](https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/docs/image_apis.md#how-to-find-mesh-names). 
You should be able to set object IDs (`simSetSegmentationObjectID`) according to your needs in the files linked in the above sections. 

This sentence is useful : `Notice that typical Unreal environments, like Blocks, usually have many other meshes that comprises of same object, for example, "Ground_2", "Ground_3" and so on. As it is tedious to set object ID for all of these meshes, AirSim also supports regular expressions.`
Instead of a regex, you'd have to manually set IDs / get the IDs and check the color values [here](https://github.com/Microsoft/AirSim/blob/42d4d91e7a7a7f037036c98c31b7537989f2a6e6/docs/seg_rgbs.txt)

1453042287, 2018-11-20:
@madratman thank you for your reply!

closed 2018-12-06 by madratman

chongjeelee, 2022-02-11:
How can i get instance segmentation label in lidar ?