microsoft/AirSim #1557: How to record RGB image?

hoangtranngoc, 2018-11-19:
In the Image API doc,  it says : "The API simGetImages can accept request for multiple image types from any cameras in single call. You can specify if image is png compressed, RGB uncompressed or float array."
So how to specify uncompressed RGB with ImageRequest() ? 
I tried another way to get uncompressed RGB by remove the last channel alpha but it seems not a correct way to do it.

bmalhigh, 2018-11-20:
@hoangtranngoc, look at the example in the Image API doc

responses = client.simGetImages([
    # png format
    airsim.ImageRequest(0, airsim.ImageType.Scene), 
    # uncompressed RGBA array bytes
    airsim.ImageRequest(1, airsim.ImageType.Scene, False, False),
    # floating point uncompressed image
    airsim.ImageRequest(1, airsim.ImageType.DepthPlanner, True)])

Also, take a look at https://github.com/Microsoft/AirSim/blob/master/AirLib/include/common/ImageCaptureBase.hpp

ImageRequest(const std::string& camera_name_val, ImageCaptureBase::ImageType image_type_val, bool pixels_as_float_val = false, bool compress_val = true)
        {
            camera_name = camera_name_val;
            image_type = image_type_val;
            pixels_as_float = pixels_as_float_val;
            compress = compress_val;
        }

hoangtranngoc, 2018-11-21:
@bmalhigh 
I had looked into examples in the Image API doc. However, my question is about getting uncompressed RGB array bytes, not uncompressed RGBA array bytes. It is because the model I trained receives input as RGB data, not RGBA. Now I need to test my navigation model by flying drone realtime in AirSim. That is why I am looking for a way to get RGB directly from airsim.ImageRequest() API. Otherwise, I need to find a proper way convert from RGBA to RGB.

Baldins, 2019-07-25:
Did you solve the issue? I have the same problem

hoangtranngoc, 2019-07-25:
Hi @Baldins ! The code below works for my case with size of image is 512x512
`responses = client.simGetImages([airsim.ImageRequest(0, airsim.ImageType.Scene, False, False)])
response = responses[0]
img1d_rgba = np.frombuffer(response.image_data_uint8, dtype=np.uint8)
img4d_rgba = img1d_rgba.reshape(512, 512, 4)
img3d_rbg = img4d_rgba[:, :, :3] `

FeilongWu, 2020-07-11:
Hi @hoangtranngoc , could you please tell me how did you run the code? I tried to take an image with the API call and I used two of my laptops, but the program was not responding for a while. One of my laptops has NVIDIA GeForce RTX 2060 GPU, which I just bought for running this program, but I am not sure if the hardware is still low to run UE4. I try to execute the code both line by line through the python command line and run them as a script, but all cases failed as the UE4 gets stuck. Any help will be appreciated! Thanks.

hoangtranngoc, 2020-07-12:
Hi @FeilongWu, RTX2060 is good to run. Is it possible that you run the example script https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py ?

closed 2022-03-31 by jonyMarino