microsoft/AirSim #1559: Multiple AirSim instances on windows server

marcociara379, 2018-11-20:
Hi devs,
first of all, thanks for the great effort.
I would like to setup a Windows server machine to run Unreal Engine and AirSim on. Ideally, me and other people I work with, would like to connect to this windows machine and create various instances of the same UnrealEngine project running on different ports using the PythonClient. 

Is there a way to do it?

Thanks

bybunni, 2018-11-21:
The issue you will run into is that the Python client talks to the game instance over a port. What you would need is for each instance to spawn on a different port. I think the easiest way to accomplish this would be to have the port number be a [command line option](https://answers.unrealengine.com/questions/123559/can-we-make-command-line-arguments.html) or part of the settings.json and have the ability to specify a particular settings.json as a command line option.

madratman, 2019-02-19:
let's track this in #1711. there's a hacky solution here https://github.com/Microsoft/AirSim/issues/1711#issuecomment-454449632

closed 2019-02-19 by madratman