microsoft/AirSim #1561: Automating Data Collection

james-mchugh, 2018-11-21:
Hello all,

Finally got up and running with AirSim today, and I am pretty excited about the segmentation feature built in. I am working on a project where I would like a drone to be able to recognize when it is above a sidewalk. To do this, I will probably gather some training data and train an network to perform image segmentation. Is there anyway to automate this type of data collection with AirSim? I know there are publicly available datasets of streets out there for image segmentation, particularly CityScapes comes to mind, but they are all taken from the perspective of a car on a street. Because of this, the domain is a bit different and they would not work for my use case.

msb336, 2018-12-14:
@james-mchugh 
Airsim has a built in segmentation engine, which allows a ground truth return image based on mesh objects in the engine. If your sidewalks are explicit objects and not part of ground, you can pull ground segmentation data directly from image capture. To automate the process, I would run AirSim in [CV mode](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#computer-vision-mode), and build a script which moves your camera along a path predefined by you and captures images every *x* frames.

You can find some CV mode and segmentation demos [here](https://github.com/Microsoft/AirSim/tree/master/PythonClient/computer_vision)

msb336, 2019-01-08:
Closing this thread due to inactivity

closed 2019-01-08 by msb336