microsoft/AirSim #1563: UE crash on setting my own camera

BrunoXL, 2018-11-22:
If a try to set up a new camera from `settings.json`, UE crashes

Below, is my settigs.json:

	{ 
	  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
	  "SettingsVersion": 1.2,
	  "SimMode": "Multirotor",
	  "CameraDefaults":{
	    "CameraSettings": [
	      {
	      "ImageType": 0,
	      "Width": 640,
	      "Height": 480,
	      "FOV_Degrees": 90,
	      "AutoExposureSpeed": 100,
	      "AutoExposureBias": 0,
	      "AutoExposureMaxBrightness": 0.64,
	      "AutoExposureMinBrightness": 0.03,
	      "MotionBlurAmount": 0,
	      "TargetGamma": 1.0,
	      "ProjectionMode": "",
	      "OrthoWidth": 5.12
	      }
	    ]
	  },
	  "Vehicles": {
	    "SimpleFlight": {
	      "VehicleType":"SimpleFlight",
	      "DefaultVehicleName": "Armed",
	      "AutoCreate": true,
	      "PawnPaths": "",
	      "EnableCollisionsPassthrough": false,
	      "EnableCollisions": true,
	      "AllowAPIAlways": true,
	      "RC":{
	        "RemoteControlID": 0,
	        "AllowAPIWhenDisconnectd": false
	      },
	    "Retrieve_Images": {
	      "VehicleType":"ComputerVision",
	      "DefaultVehicleName": "Armed",
	      "AutoCreate": true,
	      "PawnPaths": "",
	      "EnableCollisionsPassthrough": false,
	      "EnableCollisions": true,
	      "AllowAPIAlways": true,
	      "RC":{
	        "RemoteControlID": 0,
	        "AllowAPIWhenDisconnectd": false
	      },
	      "Cameras" :{
	      	"bottom-center": {
	      		"CaptureSettings": [
	      			{
	      				"ImageType" : 0,
	      				"FOV_Degrees": 120
	      			}
	      		]
	      	}
	      },
	      "Pitch": 0, "Roll": 0, "Yaw": -90 
	    }  
	  }
	}

I'm using AirSim latest code (1.2.1) and Ubuntu 16.04.

madratman, 2018-12-07:
Can you share the output of `unreal_projects/Saved/Logs`?

madratman, 2018-12-19:
@BrunoXL ?