microsoft/AirSim #1565: How to enable collision in in CV mode?

karimmamer, 2018-11-23:
Hi,

I am trying to detect collision in CV mode however the camera passes through the objects, collision count doesn't appear on Unreal and has_collided is always False. I am using the following code:
`import airsim

client = airsim.VehicleClient()
client.confirmConnection()
client.reset()

for i in range(40):
	client.simSetVehiclePose(airsim.Pose(airsim.Vector3r(i, 0, 0), airsim.to_quaternion(0, 0, 0)), False)
	info = client.simGetCollisionInfo()
	if info.has_collided:
		print(info)`
and the following settings:
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "ViewMode": "Fpv"
}`
I work with python 3, Unreal 4.18 and Blocks environment on Ubuntu 16.04.

BrunoXL, 2018-11-23:
I think CV mode has made to not allow collision!

karimmamer, 2018-11-23:
I assumed it is available after reading [this](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#setting-pose-in-computer-vision-mode) paragraph in the documentation.

jonyMarino, 2021-08-30:
See the ExternalPhysics Engine
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem.
Thanks!