microsoft/AirSim #1566: setup.sh fails on Ubuntu 16.04

ShettyHarapanahalli, 2018-11-23:
## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
when I run the setup.sh it fails with the below error on ubuntu 16.04 

Reading package lists... Done
+ sudo apt-get install -y clang-5.0 clang++-5.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package clang-5.0
E: Couldn't find any package by glob 'clang-5.0'
E: Couldn't find any package by regex 'clang-5.0'
E: Unable to locate package clang++-5.0
E: Couldn't find any package by glob 'clang++-5.0'
E: Couldn't find any package by regex 'clang++-5.0'

When I run the sudo for the above command, I get the below error - 
../AirSim$ sudo apt-get install -y clang-5.0 clang++-5.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package clang-5.0
E: Couldn't find any package by glob 'clang-5.0'
E: Couldn't find any package by regex 'clang-5.0'
E: Unable to locate package clang++-5.0
E: Couldn't find any package by glob 'clang++-5.0'
E: Couldn't find any package by regex 'clang++-5.0' 

How do I fix this ..

madratman, 2018-12-04:
Can you try?
```wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key | sudo apt-key add -
sudo apt-add-repository "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-5.0 main"
sudo apt-get update
sudo apt-get install -y clang-5.0 clang++-5.0```

ShettyHarapanahalli, 2018-12-04:
Thank for your comment, I had not selected the "Community-maintained free and open source software" option in the software and updates. I was able to install the same !!

closed 2018-12-04 by ShettyHarapanahalli