microsoft/AirSim #1567: can't run the released binary version under linux

1453042287, 2018-11-24:
when i run the CityEnviron.sh in ubuntu18.04 i got this error:

Increasing per-process limit of core file size to infinity.
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

any suggestions would be appreciate!
BTW, do i need to install the OpenGL?

tzimasak, 2018-11-24:
The AirSim environment is tested to work on Ubuntu 16.04 LTS with Unreal 4.18. I gave 18.04 a try and couldn't get it to work, but 16.04 worked right away!

1453042287, 2018-11-27:
@tzimasak thanks, oh my god...

1453042287, 2018-12-06:
@tzimasak em...i come back again, after change to 16.04, i download the CityEnviron.zip and extract it successfully, and i also add the path of .so in engine.zip to LD path, but when i run the CityEnviron.sh i still have the Segmentation fault (core dumped) error, is there something i missed?

jnaor, 2018-12-16:
I have the same problem with exactly the same error myssage (Ubuntu 16.04, Airsim 1.2.0 experimental binary). If anybody's found a solution I'd be grateful.

unrealwill, 2019-01-04:
Hi, I face the same problem. The archive is probably corrupted.
I can run the neighborhood version fine.

When I unzip the archive it complains about a data error inside the .pak file at around 60%. It also complain about missing header beginning of archive. 

I tried unzipping with various 7zip versions (both linux and windows) but always got a data error, though some files are still extracted which produce the above error when run.

unrealwill, 2019-01-04:
For reference these are probably the same issue : 
https://github.com/Microsoft/AirSim/issues/1548 
https://github.com/Microsoft/AirSim/issues/1553
https://github.com/Microsoft/AirSim/issues/1528
https://github.com/Microsoft/AirSim/issues/1345
https://github.com/Microsoft/AirSim/issues/998

aoltonen, 2019-05-31:
Hi, Have you solved the problem in the end? I got the same problem when testing the LandscapeMountains on my ubuntu16.04. I extracted the file with no error but could not get it to work.

Baldins, 2019-06-20:
did you solve the problem?

jnaor, 2019-06-20:
No...

ironclownfish, 2020-08-13:
We will check that our latest release of CityEnviron works on linux.

nightduck, 2020-08-26:
What is this engine.zip everyone is speaking of? I'm getting library errors running [CityEnviron from 1.2](https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux). It complains about a missing libopenvr_api.so file. I see other environments with an Engine folder that has that library, but if I borrow their copy and add it to LD path, I get the signal 11 segfault that OP mentioned.

@ironclownfish Is the "latest release" of CityEnviron also going to take out the cars and pedestrians, like in the Windows 1.3.1 release?

Also using UE 4.24 on Ubuntu 18.04

qiujiangkun, 2020-08-31:
I compiled the version from source code and have the same problem, too.
I cannot run `./UnrealVersionSelector-Linux-Shipping`, with or without `-opengl4` option. But I can run UE4Editor without crashing.
```log
$ screenfetch
 ██████████████████  ████████     OS: Manjaro 20.1 Mikah
 ██████████████████  ████████     Kernel: x86_64 Linux 5.4.57-1-MANJARO
 ██████████████████  ████████     Uptime: 11h 8m
 ████████            ████████     Packages: 1463
 ████████  ████████  ████████     Shell: bash 5.0.18
 ████████  ████████  ████████     Resolution: 1920x1080
 ████████  ████████  ████████     DE: Xfce4
 ████████  ████████  ████████     WM: Xfwm4
 ████████  ████████  ████████     WM Theme: Matcha-sea
 ████████  ████████  ████████     GTK Theme: Matcha-sea [GTK2]
 ████████  ████████  ████████     Icon Theme: Papirus-Maia
 ████████  ████████  ████████     Font: Noto Sans 10
 ████████  ████████  ████████     Disk: 174G / 1.2T (16%)
                                  CPU: Intel Core i5-4590 @ 4x 3.7GHz [44.0°C]
                                  GPU: GeForce GTX 1660
                                  RAM: 2593MiB / 7901MiB

```

```log
git log
commit df84cb430f38ad08ad831f31267d8702b2fefc3e (grafted, HEAD -> release, tag: 4.25.3-release, origin/release, origin/HEAD)
Author: unrealbot <unrealbot@epicgames.com>
Date:   Tue Jul 28 15:06:13 2020 -0400
```

nightduck, 2020-08-31:
Are you talking about the City environment? Or the unreal engine itself? Code is not available for the former

qiujiangkun, 2020-08-31:
> Are you talking about the City environment? Or the unreal engine itself? Code is not available for the former

Oh I mean the engine itself. Others may mean City environment

ElMonfo00, 2024-05-10:
> What is this engine.zip everyone is speaking of? I'm getting library errors running [CityEnviron from 1.2](https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux). It complains about a missing libopenvr_api.so file. I see other environments with an Engine folder that has that library, but if I borrow their copy and add it to LD path, I get the signal 11 segfault that OP mentioned.
> 
> @ironclownfish Is the "latest release" of CityEnviron also going to take out the cars and pedestrians, like in the Windows 1.3.1 release?
> 
> Also using UE 4.24 on Ubuntu 18.04.
> 
> Update: CityEnviron binaries also segfault on 16.04

At the end have you solved the problem? Or is it not possible install CityEnviron in linux?

aoltonen, 2024-05-10:
您好，邮件已收到，谢谢！