microsoft/AirSim #1570: cmake error with VS2017 when building AirSim

paradox56, 2018-11-26:
Hello, I try to build AirSim with Visual Studio 2017. I cloned the repository and ran `build.cmd`

I have the following error
```
Found cmake version: 3.12.18081601-MSVC_2
Starting cmake to build rpclib...
CMake Error: The source directory "C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Error: C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1/build is not a directory
Error: C:/Windows/System32/AirSim/external/rpclib/rpclib-2.2.1/build is not a directory

#### Build failed - see messages above.
```

paradox56, 2018-11-26:
The error disappeared after I moved the AirSim folder to D:/

closed 2018-11-26 by paradox56

guiarreche, 2020-05-31:
It worked for me too, I just moved the folder to the desktop! Thanks