microsoft/AirSim #1572: Still getting the same resolution images 256x144 with json settings as below.

chaitanyantr, 2018-11-26:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

chaitanyantr, 2018-11-26:
Airsim - 1.2
windows 10
unreal engine 4.18
land scape mountains - env
[image]

{
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
      {
        "CameraName": "0",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      }
    ]
  },
  "CaptureSettings": [
    {
      "ImageType": 0,
      "Width": 1280,
      "Height": 720,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100,
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    }
  ]
}

chaitanyantr, 2018-11-27:
forgot to put "CameraDefaults": { } in previous json.

Now it is fine working...Resolved !

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",

  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 1,
        "Width": 640,
        "Height": 480
      },

      {
        "ImageType": 2,
        "Width": 640,
        "Height": 480
      },

      {
        "ImageType": 0,
        "Width": 640,
        "Height": 480
      }

    ]

  }
}

closed 2018-11-27 by chaitanyantr

keerats95, 2019-11-11:
I am having the same problem. 
Here is my json file

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",

  "Recording": {
    "RecordOnMove": false,
    "Cameras": [
        { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true }
    ]
  },
  "CameraDefaults": {
    "Gimbal": {
      "Stabilization": 1,
      "Pitch": 1, "Roll": 1, "Yaw": 0
    },
	"CaptureSettings": [
        {
          "ImageType": 2,
          "Width": 1280,
          "Height": 960
        }
    ],	
    "X": 1,
    "Y": 1,
    "Z": 1,
    "Pitch": 1,
    "Roll": 1,
    "Yaw": 0
  }
}

chaitanyantr, 2019-11-12:
Can you paste this .json file and check if the resolution is changing for you.?

{
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
"SettingsVersion": 1.2,
"SimMode": "Multirotor",

"CameraDefaults": {
"CaptureSettings": [
{
"ImageType": 1,
"Width": 640,
"Height": 480
},

  {
    "ImageType": 2,
    "Width": 640,
    "Height": 480
  },

  {
    "ImageType": 0,
    "Width": 640,
    "Height": 480
  }

]
}
}

keerats95, 2019-11-13:
Hi,

Thanks for the input.
I tried this already.
I was having a problem where my picture being saved in a lower resolution
no matter what. I was using matplotlib to save the file from python and had
to change the dpi to get a clearer image.

Another thing, I was wanting to kill the yaw of the controller. Could I do
this from the .json file?

Keerat Singh