microsoft/AirSim #1573: Drone Shell: Arm, TakeOff, Land, Disarm works. But no other commands (e.g. MoveToPostion) seem to have any effect. No errors reported.

SciTech-Ent, 2018-11-26:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

SciTech-Ent, 2018-11-26:
Hi All,
I have a problem which I suspect is a settings problem: 
I am using Drone Shell: Arm, TakeOff, Land, Disarm works, however, no other commands (e.g. MoveToPostion -x 100) seem to have any effect once airborne.
No errors are reported.
I am not using Hardware-in-the-loop.
Behaviour is the same in both Unreal and Unity.

Any help will be appreciated, as I am now completely stuck.

Thanks

SciTech-Ent, 2018-11-26:
@lovettchris Hi Chris, any advice on the issue above? I am sure I am not the only one experiencing this problem. Any suggestions from anyone?

madratman, 2019-02-19:
closing coz of lack of activity

closed 2019-02-19 by madratman