microsoft/AirSim #1575: BaroMeter and GPS reset

yashkaushal, 2018-11-27:
Hi,

I monitor the movement of the multirotor using QGroundControl in the HITL mode using PX4 on PixHawk, and after every 10-15 seconds the GPS position is reset to home location(microsoft's office) and the readings from barometer stay in the range of -2 to 10m. PixHawk shows the GPS failure and Barometer glich LEDs on different modes.

Sometimes errors are displayed in AirSim in green text with no apparent pattern to check sensor calibration and navigation failure.

If i make the PX4 take off using QGC, it climbs indefinitely and according to barometer reading it never reaches the required altitude.

Using simpleflight the GPS and altitude readings are not updated at all (monitored using getmultirotor state API)

The drone however flies well in both SimpleFlight using APIs and HITL in manual and stabilised mode using RC.

I have used the source of 14-nov-2018 to build and tried using the prebuilt binaries too,

System specs - i7 4770, 8 GB RAM(around 2gb free while running airsim) and samsung 840 EVO SSD
Airsim version 1.2, windows 10, python 2.7.15, unreal engine 4.20, PX4 firmware 1.8.1

yashkaushal, 2018-12-05:
I found in documentation that SimpleFlight is not expected to have GPS and Barometer readings, so moving on with HITL I did the following.

I tested that with version 1.1 of AirSim and the bug was not there then after going through the source code of AirSim, I commented below two lines in `PawnSimAPI::update()` in `PawnSimApi.cpp`
```
environment_->update();
kinematics_->update();
```
and added  below two lines at the end of `updatePhysics(PhysicsBody& body)` in `FastPhysicsEngine.hpp`
```
body.getEnvironment().setPosition(next.pose.position);
body.getEnvironment().update();
```

Recompiling after these changes and now the GPS and the Barometer reading are working fine in HITL(tested with multiple HITL as well) and in SITL(single instance).

Shall I create a pull request to add those changes in AirSim on GitHub?

mikeisted, 2018-12-07:
Thanks very much for that - very helpful indeed.

In the short term we have had to postpone plans to use AirSim because of this issue.  With some luck we may be able to revisit the situation in the new year and will certainly try this fix.

yashkaushal, 2018-12-10:
You're welcome Mike, glad this could help you.

lovettchris, 2018-12-18:
There were bugs in the multirotor simulator class causing confusion over duplication of state.  But these problems should all be fixed now.  I've tested that latest AirSim bits work fine with PX4 version 1.8.2.

mikeisted, 2018-12-19:
Thank you Chris,
I look forward to trying again in the New Year!
Mike

closed 2018-12-19 by madratman