microsoft/AirSim #1576: use turtlebot model to replace car

jianghe01, 2018-11-27:
Hi, i want to use turtlebot model to replace the car model to run airsim.
first i created a turtlebot fbx model by maya and import it to unreal. i set the physics bodies of bones to kinematic type to prevent them dividing.
secondly i created tire data and animation blueprint.
then i created a wheeledvehicle blueprint class inherited from carpawn. i set the class as you say in the PawnPaths settings and change the PawnPaths settings.
i also modified the input of project settings.

now it can render the turtlebot model to replace the car model. but i cannot move it by keyboard or program. can anyone help me to do it correctly. thanks a lot

RVBldr, 2018-12-19:
@jianghe01 This is an Unreal question, so the best way forward is to review Unreal documentation on vehicle creation, as well as the many YouTube videos on creating actors for Unreal. That should help you with the control mappings.

shaoxiang, 2020-06-01:
@jianghe01 Have you solved the problem? I also want to use robots in airsim.

jianghe01, 2020-06-02:
No, I have not solved this problem. Sorry I cannot give you any more advice because I’m not working on the project now.

shaoxiang, 2020-06-03:
@jianghe01  It's sad to hear that this problem can't be solved in airsim...

jonyMarino, 2020-09-02:
related to #1506

zhoudacheng6, 2021-03-01:
Have someone solved the problem? I also want to use robots in airsim.

jonyMarino, 2022-04-19:
Take a look at the urdfAirSim repository for examples