microsoft/AirSim #1577: UE4Editor-AirSim.dll Error when trying to build Unreal Project

Dola47, 2018-11-27:
Hello,

I am trying to create a Unreal project. I am following the steps provided in this video [https://www.youtube.com/watch?v=1oY8Qu5maQQ](https://www.youtube.com/watch?v=1oY8Qu5maQQ) 

I have done all the steps after merging MyProject with Landscape Mountains and trying to reopen my project I always get this error:
[image]
[image] .

I do not know what is the reason for that.

I cloned AirSim normally, I adjusted all the projects to run with SDK Windows 10.0.17763.132 and AirSim built successfully, also SDK 8.1 is installed in VisualStudio Community 2017, I copied Plugins and pasted it in Myproject folder, I have done all the steps exactly like the video.

Any suggestions what may be the reason for that.

Thanks.

msb336, 2018-11-27:
@Dola47 Which version of Unreal Engine are you using? Have you been able to successfully get the ["Blocks" trial environment](https://microsoft.github.io/AirSim/docs/unreal_blocks/) running with the airsim plugin?
Currently airsim is only compatible with Unreal Engine v 4.18.

Dola47, 2018-11-27:
@msb336 I was using Unreal Engine 4.14.3 the same version as in the Youtube, and it worked successfully with him. I have not tried "Blocks" and I followed the steps in the video and I think it should work. 
I can give it a try for Unreal Engine v 4.18 also.

Dola47, 2018-11-28:
@msb336 Thanks right now when I installed version 4.18 it worked but I have a problem that the drone does not fly. However, the car move successfully did you know what may be the reason for that?

bmalhigh, 2018-11-28:
Since you opened #1580 on drone flying, we should close this. Thanks.

closed 2018-11-29 by bmalhigh