microsoft/AirSim #1578: How to modify quadcopter's parameters

MoritzGa, 2018-11-27:
Hello AirSim community,

I'm trying to **adjust AirSim** in such a way that it **represents the real quadcopter**. For that I would have to modify the hardcoded parameters in **("AirLib/include/vehicles/multirotor/firmwares/simple_flight/SimpleFlightQuadXParams.hpp")** for example? Am I missing a way to tune them by using the **json file** or do I have to work on my own branch and modify them within the params file?

Thanks in advance.

madratman, 2018-12-19:
Hi @MoritzGa, good question!
While, this is currently not supported, here's a quick jab at how to get it done. 

- The main work is done by [MultiRotorParams:initialize()](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp#L59-L67). You can actually see that sensors are being added grabbing the sensor settings from `vehicle_settings`. So, let's try to imitate the way sensors are being grabbed from vehicle setting.

- So, first, you'll need to add a struct called something like `RotorParamsSettings` in [AirSimSettings.hpp](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/common/AirSimSettings.hpp), matching the actual [MultiRotorParams::Params struct](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/MultiRotorParams.hpp#L30) being filled in by [SimpleFlightQuadXParams::setupParams()](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/simple_flight/SimpleFlightQuadXParams.hpp). 
(just look at how camera/lidar/sensor settings are being added in that file)

Then you'll add this `MultiRotorParams` you made as a member of the [VehicleSettings](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/common/AirSimSettings.hpp#L223) struct in the way cameras and sensors are added. (`std::map<std::string, RotorParamsSettings> multirotorparams`). 
And also a load function on the lines of [this](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/common/AirSimSettings.hpp#L720). 

Finally you can change setupParams() in SimpleFlightQuadXParams,h and get the rotor settings frmo the `vehicle_setting_` object which is initialized in the constructor. 

Let me know if you have more questions. If you get it working, feel free to send in a PR!

closed 2019-01-08 by madratman

MoritzGa, 2019-01-21:
Thanks for the answer @madratman.

I tried implementing the proposed solution but I ran into issues. I added the following code to AirSimSettings.hpp:

	struct MultiRotorParamsSettings {
		//required
		uint rotor_count;
		vector<msr::airlib::MultiRotorParams::RotorPose> rotor_poses;
		real_T mass;
		Matrix3x3r inertia;
		Vector3r body_box;

		////optional
		real_T linear_drag_coefficient = 1.3f / 4.0f;
		//sample value 1.3 from http://klsin.bpmsg.com/how-fast-can-a-quadcopter-fly/, but divided by 4 to account
		// for nice streamlined frame design and allow higher top speed which is more fun.
		//angular coefficient is usually 10X smaller than linear, however we should replace this with exact number
		//http://physics.stackexchange.com/q/304742/14061
		real_T angular_drag_coefficient = linear_drag_coefficient;
		real_T restitution = 0.55f; // value of 1 would result in perfectly elastic collisions, 0 would be completely inelastic.
		real_T friction = 0.5f;
		msr::airlib::RotorParams rotor_params;
	};

and added `#include "vehicles/multirotor/MultiRotorParams.hpp"`. When building it I get tons of erros, I believe there is something wrong with the include. Do I have to modify the respective CMakeLists.txt?