microsoft/AirSim #1579: Unity - Confusion about where settings.json file should be located?

jasonhbartlett, 2018-11-27:
Win 10 Pro
Unity 2018.2.14f1 - Running in editor (hit Play)
VS Community 2017 15.8.8
SDK 8.1

ref: [https://github.com/Microsoft/AirSim/issues/1550](url)

Although it was pointed out in issue #1550 that the settings.json file needed to be placed in the C:\Program Files\Unity\Editor folder, I am confused as to why this is the case?  Having no settings.json file there causes the error "Problem in starting AirSim server!!! Please check logs for more information." and kills the Unity editor.  This is not the behavior AirSim typically has as typically the file is created automatically if it is missing, albiet it is typically created in the Documents/AirSim folder.

Also if not getting an error (cause if found the file in the editor folder) it WILL STILL create a settings.json file in the Documents/AirSim folder.  And I'm led to suspect that this is actually where the project settings are pulled from?

So, where exactly is the settings.json file needed?  And where are the settings pulled from?

msb336, 2018-11-27:
Unity calls for settings.json inside "InitializeAirSim.cs", which is located here:
Unity\UnityDemo\Assets\AirSimAssets\Scripts

```
    public static string GetAirSimSettingsFileName()
    {
        string fileName = Application.dataPath + "\\..\\settings.json";
        if (File.Exists(fileName))
        {
            return fileName;
        }

        fileName = Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.MyDocuments), Path.Combine("AirSim", "settings.json"));
        if (File.Exists(fileName))
        {
            return fileName;
        }

        if (CreateSettingsFileWithDefaultValues(fileName))
            return fileName;
        else
            return string.Empty;
    }
```

Unity looks for settings.json inside of Documents\Airsim inside of the project directory. If it can't find either it will build a default  settings.json file inside of Documents\Airsim.

jasonhbartlett, 2018-12-10:
@msb336 Just poking around.  I believe the error message in the Unity console is coming from:
AirSim/Unity/AirLibWrapper/AirsimWrapper/Source/SimHUD/SimHUD.cpp Line 147.
This section is where it tries to read the file from the same location as the Unity.exe Editor executable.  Don't know why we are looking there for the settings file?  A.) We already are checking this in the Unity scripts and B.) Should only be looking in Documents\AirSim.  Correct?

msb336, 2018-12-10:
@jasonhbartlett 
It looks you are correct. This must be an artifact from a pre-release version of the unity wrapper. SimHUD::getSettingsText() looks in both the executable directory and the standard airsim directory.

https://github.com/Microsoft/AirSim/blob/96b449e570d8d241dc32e1c18c0c9d53bcad4d80/Unity/AirLibWrapper/AirsimWrapper/Source/SimHUD/SimHUD.cpp#L126-L132

Once I can confirm deleting the executable path read doesn't break the build I'll add it to the PR.

Thanks!

jasonhbartlett, 2018-12-11:
@msb366  Awesome!  I'm looking forward to the merge of all new updates folks like yourself are working on.  Let me know if there is anything I can do to help.  I'm new to AirSim but have a bit of experience with Unity and I'm really hoping Unity will become as well developed as Unreal is now.  I know that might be a little ways off yet.  Anyhow, thanks for your work on this!

jasonhbartlett, 2019-01-09:
@RVBldr @msb366 
Still getting the error.  SimHUD.cpp file hasn't yet been fixed as noted above.  Not a breaking issue though.

msb336, 2019-01-10:
@jasonhbartlett 
Searching the executable directory is an intended feature, so just make sure you are careful about where you are storing your settings.json files.

jasonhbartlett, 2019-01-10:
@msb336 
Sorry, I'm not following.  Doesn't Unreal build look at Documents\AirSim\ dir for settings.json file?  Why would Unity be different?  Does this mean I'll just always see that error when running in the editor?

I would prefer to only have to worry about making changes to the settings file in one location.

msb336, 2019-01-10:
@jasonhbartlett Unreal build also looks in the executable directory. https://github.com/Microsoft/AirSim/blob/5246acd664c8cf4aeb1070c0e9ea64fdcf442ddd/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L351-L358

I agree, it is confusing to have to monitor two locations for the settings file. My recommendation would be to only have a `settings.json` file located in your documents folder. AirSim won't create a default settings file inside the executable folder.

jasonhbartlett, 2019-01-10:
FYI  There are a couple of other issues I've noticed with the AirSim Unity build.  One would be an easy fix:  The PC DroneDemo Unity project has an extra empty game object with a 2nd InitializeAirSim script component attached. Should be deleted.

[image]

Second, is not as easy fix.  I mentioned in in Issue #1617 Basically the AirSim/Unity/UnityDemo/Assets/AirSimAssets/Scripts/Utilities/AirSimSettings.cs script need to be updated to be able to read the "Vehicles" data structure.

Should I go ahead and open a new issue for this?

PS- At some point I hope to be contributor to this project.

msb336, 2019-01-10:
Yes, please open a new issue!

If you want to contribute, that would be really great! We're currently low on bandwidth for unity support, so community help will be key in bringing it up to par with the unreal side. We have some guidelines for contributing [here](https://github.com/Microsoft/AirSim/blob/master/CONTRIBUTING.md)

msb336, 2019-01-10:
If we're wrapped up with this issue, I'm going to go ahead and close it.

jasonhbartlett, 2019-01-11:
@msb336 Okay I opened new issue #1696 for that.  And we can go ahead and close this issue.  Thanks for your help and I'll get familiar with the contributing guidelines.

closed 2019-01-11 by msb336

blacknaki, 2020-05-02:
Cannot read settings file  C:\Program Files\Unity\Hub\Editor\2018.2.15f1\Editor\settings.json Vehicle=PhysXCar

I get this error. Does anyone know why?

AngelTang190, 2020-05-09:
> Cannot read settings file C:\Program Files\Unity\Hub\Editor\2018.2.15f1\Editor\settings.json Vehicle=PhysXCar
> 
> I get this error. Does anyone know why?

Me too!

blacknaki, 2020-05-19:
any solution yet?

AngelTang190, 2020-05-19:
Not sure though

jasonhbartlett, 2020-05-19:
Sorry I've gone off this project for a long time.  Maybe @msb336 will be able to help.

mcaisw, 2023-12-29:
Cannot read settings file  D:\Program Files\Unity\2022.3.8f1c1\Editor\settings.json Vehicle=    
I also got this error, it's almost 2024