microsoft/AirSim #1580: Drone does not fly

Dola47, 2018-11-28:
Hello,
 I have a problem that my drone does not fly in both Unity and Unreal. However, my car moves correctly.

I have my AirSim build correctly and everything is working fine. The only issue that I do not know how to let my Drone fly using the keyboard. When I press page up + arrow key or WASD nothing happens. 

Thanks.

Dola47, 2018-11-28:
It turns out the I have an error while trying to compile HelloDrone project that IntelliSense may be missing for this module. 

Any suggestions how to overcome this error. 

[image]

Dola47, 2018-11-28:
I checked HelloDrone project and it builds successfully without errors, so it is expected that the drone can fly. I wonder why it is not flying when I press arrow keys or WASD + page up button.

Can any one specify exactly how he was able to let the drone fly using the keyboard.

Thanks.

bmalhigh, 2018-11-29:
I don't think you can fly drone with keyboard. Use the APIs or use a remote control (e.g., XBox controller)

Dola47, 2018-11-29:
@bmalhigh  Thanks for your response. 

What I understand that HelloDrone  is an already working example and has a specified API within it is code, and so the Drone has to fly.

Otherwise,  I got this wrong, and then I should define the API myself in HelloDrone.

Please, clarify this to me

jasonhbartlett, 2018-11-29:
Hi @Dola47 
To fly via API using the Python client for example, (of course python must be installed and set on $PATH variable), you would use command window to navigate to the AirSim/PythonClient/multirotor directory and then try out one of the python scripts there.  For example ...\multirotor>python orbit.py

Make sure to start the sim running in Unity first though.  The drone should start flying based on the code in the python script.

Dola47, 2018-11-29:
@jasonhbartlett 

Can you state in a more detailed way how to set on $PATH variable because I have not got it. 

Moreover, If I want to get everything done in C++, then I see that there is only HelloDrone that I can use for moving my drone, so final word does HelloDrone will able me any way to fly my Drone or to fly I should use the python files instead. 

Thanks a lot

jasonhbartlett, 2018-11-29:
@Dola47 - Sorry, I'm not sure about the C++ question.  I'm pretty new to this project and not up to speed on C++ much.  Probably that answer is yes, that you can control the drone with C++ scripts like HelloDrone.
As for adding python to Windows $PATH variable - [http://lmgtfy.com/?q=windows+add+python+to+path](url) 

You can also run those by opening the VS solution file for the PythonClient, pick one of the scripts as the startup file and hit F5.

Also, FYI - they previously included code in the Drone.cs script which allowed for flying manually with an Xbox controller, but they pulled it out for some reason.  I kind of liked using it so I made a DroneControllerInput.cs script that can be attached to the Quadcoptor game object which has the Drone.cs script in the DroneDemo scene.

Unfortunately it still requires to add 2 public methods to the Vehicle.cs:

`        // Added for DroneControllerInput Use
        public bool ControllerAllowed() {
            return isServerStarted && !isApiEnabled;
        }

        // Added for DroneControllerInput Use
        public void SetRCData(AirSimRCData data) {
            rcData = data;
        }`

The DroneControllerInput.cs looks like this:

`using System;
using System.Collections.Generic;
using AirSimUnity.DroneStructs;
using UnityEngine;

namespace AirSimUnity
{
    [RequireComponent(typeof(Drone))]
    public class DroneControllerInput : MonoBehaviour
    {
        private Drone drone;

        private void Start() {
            drone = this.gameObject.GetComponent<Drone>();
        }
        private void LateUpdate()
        {
            if (drone.ControllerAllowed())
            {
                AirSimRCData rcData = drone.GetRCData();

                rcData.is_valid = true;
                rcData.roll = Input.GetAxis("Horizontal");
                rcData.pitch = Input.GetAxis("Vertical");
                rcData.throttle = Input.GetAxis("Depth");
                rcData.yaw = Input.GetAxis("Yaw");
                rcData.left_z = 0;
                rcData.right_z = 0;

                rcData.switch1 = (uint)(Input.GetKeyDown("joystick button 0") ? 1 : 0);
                rcData.switch2 = (uint)(Input.GetKeyDown("joystick button 1") ? 1 : 0);
                rcData.switch3 = (uint)(Input.GetKeyDown("joystick button 2") ? 1 : 0);
                rcData.switch4 = (uint)(Input.GetKeyDown("joystick button 3") ? 1 : 0);
                rcData.switch5 = (uint)(Input.GetKeyDown("joystick button 4") ? 1 : 0);
                rcData.switch6 = (uint)(Input.GetKeyDown("joystick button 5") ? 1 : 0);
                rcData.switch7 = (uint)(Input.GetKeyDown("joystick button 6") ? 1 : 0);
                rcData.switch8 = (uint)(Input.GetKeyDown("joystick button 7") ? 1 : 0);

                drone.SetRCData(rcData);
            }
        }
    }
}`

Lastly, you need to update the InputManager.asset which is in the Unity ProjectSettings folder.  This is what mine looks like:

`%YAML 1.1
%TAG !u! tag:unity3d.com,2011:
--- !u!13 &1
[... 332 lines cut ...]
    gravity: 0
    dead: 0.19
    sensitivity: 1
    snap: 0
    invert: 0
    type: 2
    axis: 0
    joyNum: 0
  - serializedVersion: 3
    m_Name: Depth
    descriptiveName: 
    descriptiveNegativeName: 
    negativeButton: 
    positiveButton: 
    altNegativeButton: 
    altPositiveButton: 
    gravity: 0
    dead: 0.19
    sensitivity: 1
    snap: 0
    invert: 1
    type: 2
    axis: 1
    joyNum: 0
`

closed 2020-09-09 by jonyMarino