microsoft/AirSim #1583: log_messages_hidden bug: code doesn't clear screen

Lotjeotje, 2018-11-28:
In the LogMessage function in AirBlueprintLib.cpp (line 155 - 159) there is a bug in the part where log_messages_hidden is set to true. It only enables printing new LogMessages (so updating it) but the code doesn't clear the screen. 

```
void UAirBlueprintLib::LogMessage(const FString &prefix, const FString &suffix, LogDebugLevel level, float persist_sec)
{
    if (log_messages_hidden_)
        return;

...
```
I solved the problem by writing is like this: 

```
if (log_messages_hidden_) 
{
	if (GEngine) {
		GEngine->ClearOnScreenDebugMessages();
	}
	return;
}

```

NextSim, 2018-12-03:
#1007

NextSim, 2018-12-03:
#1007

madratman, 2019-01-23:
Let's track this in #1007. Closing to avoid duplicates, we already have too many issues

closed 2019-01-23 by madratman