microsoft/AirSim #1585: Question/bug: NED offset in linear velocity (wrong use of toLocalNed() in updateKinematics())

Lotjeotje, 2018-11-28:
In the updateKinematics function in PawnSimApi (line 485 - 500), the linear twist is being transformed using the toLocalNed function from NedTransform.cpp (line 29 - 34). This function subtracts a ned_offset from the position, but in the updateKinematics function, the velocity is the input to toLocalNed(position). Since I have an Ned offset of {X=5690.00000 Y=-350.000000 Z=18.9090881 }, my linear velocity is having this offset (scaled with 1/100) when playing the game. 

Question: I'm trying to fix this by having the originGeoPoint and PlayerStart in the settings such that there is no ned offset in the code now. Is there any other way that I can get a correct linear velocity? Maybe making a toLocalNed function for transforming velocity (only scaling and not translating?).

```
void PawnSimApi::updateKinematics(float dt)
{
    const auto last_kinematics = kinematics_;

    kinematics_.pose = getPose();

    kinematics_.twist.linear = getNedTransform().toLocalNed(getPawn()->GetVelocity());
    kinematics_.twist.angular = msr::airlib::VectorMath::toAngularVelocity(
        kinematics_.pose.orientation, last_kinematics.pose.orientation, dt);

    kinematics_.accelerations.linear = (kinematics_.twist.linear - last_kinematics.twist.linear) / dt;
    kinematics_.accelerations.angular = (kinematics_.twist.angular - last_kinematics.twist.angular) / dt;

    //TODO: update other fields?

}
```

```
NedTransform::Vector3r NedTransform::toLocalNed(const FVector& position) const
{
    return NedTransform::toVector3r(position - local_ned_offset_,
        1 / world_to_meters_, true);
}
```

Lotjeotje, 2018-11-28:
I solved it by using this toLocalNedVel(const FVector& velocity) function in the line where kinematics_.twist.linear is updated.

```
NedTransform::Vector3r NedTransform::toLocalNedVel(const FVector& velocity) const
{
	return NedTransform::toVector3r(velocity,
		1 / world_to_meters_, true);
}
```

ironclownfish, 2020-05-19:
May be relevant:

https://github.com/microsoft/AirSim/pull/2292
https://github.com/microsoft/AirSim/pull/2299

closed 2020-05-19 by madratman