microsoft/AirSim #1589: Error building AirLibWrapper for Unity

zippopalermo, 2018-11-29:
`simmodebase.cpp(142): error C2660: 'msr::airlib::ApiServerBase::start': function does not take 0 arguments [AirsimWrapper
.vcxproj]`

The api_server start function is currently expecting two arguments (ApiServerBase.hpp line 14):

`virtual void start(bool block, std::size_t thread_count) = 0;`

It's currently being called with no arguments passed from simmodebase.cpp line 142:

`api_server_->start();`

kenbier, 2018-11-29:
Looks like a change to the Airlib library is breaking the AirLib wrapper. We had CI testing both libraries but it looks like Unity testing has been turned off. Understandably a broken change was merged since we are flying blind.

I will work on a fix and turning on Unity testing so this doesn't happen in the future. In the meantime can you delete issue #1588 and add those details to this issue? Thanks.

zippopalermo, 2018-11-29:
Details from Issue #1588 as stated above:
When trying to build the AirsimWrapper Plugin there's currently an error in simmodebase.cpp while calling reset on WorldSimApi. An argument is being passed to the reset function on line 18:

world_sim_api_.reset(new WorldSimApi(this, vehicle_name_));

The reset function in WorldSimApi currently doesn't take any arguments, though (WorldSimApi.cpp line 18):

void WorldSimApi::reset() { simmode_->reset(); }

kenbier, 2018-12-04:
@zippopalermo I believe this was fix by https://github.com/Microsoft/AirSim/commit/6eeba15b5d76c4b90995ba64ebbeea9b3fa40709. Can you try building again?

zippopalermo, 2018-12-05:
Looking good. No problems building.
Thanks for the fix.
=> Solved

closed 2018-12-05 by zippopalermo