microsoft/AirSim #1591: 2D Lidar: Channel missing

edualvarado, 2018-11-29:
OS: Ubuntu 16.04
AirSim 1.2

I am currently adding a Lidar sensor to my vehicle. I need a 2D Lidar, but when I change the number of channels to 1 in the .json file, the sensor won't show any points. Only when the number of channels are 2 or more it works.

The position with respect to the vehicle seems fine to me. As I said, it works perfectly for 2 or more channels, being the points shown where I want.

This is my current configuration for the Lidar:

      "Lidar1": {
         "SensorType": 6,
         "Enabled" : true,
         "NumberOfChannels": 1,
         "RotationsPerSecond": 10,
         "PointsPerSecond": 10000,
         "X": 0, "Y": 0, "Z": -0.15,
         "Roll": 0, "Pitch": 0, "Yaw" : 0,
         "VerticalFOVUpper": 0,
         "VerticalFOVLower": 0,
         "HorizontalFOVStart": 0,
         "HorizontalFOVEnd": 0,
         "DrawDebugPoints": true
       }

bmalhigh, 2018-11-29:
Investigating.

closed 2018-11-29 by sytelus

bmalhigh, 2018-11-30:
@edualvarado -- Thanks much for reporting the issue. It is fixed now.

PR #1593 (2018-11-29): Fix a bug in Lidar for single channel scenario.
Also fixed couple of compiler warnings occurring in Ubuntu.

Resolves: #1591
commit b69759f3 (2018-11-29): Fix a bug in Lidar for one channel scenario.

Also fixed couple of compiler warnings occuring in Ubuntu.

Resolves: #1591