microsoft/AirSim #1594: Block build error by UE4.18  on Ubuntu16.04

yidongVSI, 2018-11-29:
Block build failed by UE4.18 on Ubuntu 16.04
The AirSim is in 
`SHA 65f45b740ddabf9090b06f9dd23397dd61f4b064`

Most of compile errors were from 
`AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/Weather/WeatherLib.h`

as below
`WeatherLib.h:55:10: error: no viable conversion from returned value of type 'const wchar_t [62]' to function return type 'const FSoftClassPath' return TEXT("AActor'/AirSim/Weather/WeatherFX/WeatherActor.WeatherActor_C'")`

`WeatherLib.h:50:10: error: no viable conversion from returned value of type 'const wchar_t [49]' to function return type 'const FSoftClassPath' return TEXT("AActor'/AirSim/Weather/UI/MenuActor.MenuActor_C'")`

`WeatherLib.h:60:10: error: no viable conversion from returned value of type 'const wchar_t [58]' to function return type 'const FSoftClassPath' return TEXT("UUserWidget'/AirSim/Weather/UI/OptionsMenu.OptionsMenu_C'")`

seems there is some version confliction between AirSim and Unreal Engine 4.18 ??  As I built Block example, I always got a warning message before building saying 'This project was made with a difference version of the Unreal Engine.  Converting to this version will rebuild your code projects...'

Thanks for any suggestions!

bmalhigh, 2018-11-29:
@sytelus is looking at this issue.

MagneticWind, 2018-12-03:
Same issue here, I am seeing it on Ubuntu 18.04.1 LTS when building blocks project. Thanks for looking at it.

bigsheep2018, 2018-12-04:
I get similar error. I build it with QT Creator.
I get a notice asking to rebuild:
libUE4Editor-Blocks.so 
libUE4Editor-AirSim.so
click on yes. 
Then the following errors shows in QT consolo:
/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/Weather/WeatherLib.h:50:10: error: no viable conversion from returned value of type 'const wchar_t [49]' to function return type 'const FSoftClassPath' return TEXT("AActor'/AirSim/Weather/UI/MenuActor.MenuActor_C'");

/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/Weather/WeatherLib.h:55:10: error: no viable conversion from returned value of type 'const wchar_t [62]' to function return type 'const FSoftClassPath' return TEXT("AActor'/AirSim/Weather/WeatherFX/WeatherActor.WeatherActor_C'");

/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/Weather/WeatherLib.h:60:10: error: no viable conversion from returned value of type 'const wchar_t [58]' to function return type 'const FSoftClassPath' return TEXT("UUserWidget'/AirSim/Weather/UI/OptionsMenu.OptionsMenu_C'");

KevinLADLee, 2018-12-05:
Same issue here.
Ubuntu 16.04.5 LTS with UE 4.18.2 when building blocks project.

yipstar, 2018-12-06:
Having the same issue as well.
Ubuntu 16.04.3 LTS
UE 4.18
AirSm v1.2.0Linux

madratman, 2020-08-11:
closing as outdated wrt current master

closed 2020-08-11 by madratman