microsoft/AirSim #1595: MSB1009 error

Ahmad5112, 2018-11-30:
I am trying to build AirSim  I have gone through all the steps described to build but all  the time  I got the same error  and I am unable to resolve the issue if anybody there  can help me to resolve 
[image]

bmalhigh, 2018-11-30:
You are building this under \windows\system32 ? Wondering if you are running into some perms issue.

Also, did you check what is in 
\external\rpclib\rpclib-2.2.1\build\CMakeFiles\CMakeOutput.log
?

SongjunHuang, 2021-11-10:
Hi I wanna know which folder do you mean and what permissions should I give to this folder?