microsoft/AirSim #1596: Support for official docker files

varun19299, 2018-12-01:
* Python Version : 3.5 
* Ubuntu 16.04
* UE 4.18 (tried UE 4.19, works)

Can AirSim support a dockerfile as a part of its builds? This makes reproducibility far easier on Linux.

Also, a look at CARLA's docker file release could be helpful (also uses UE, UE 4.19). Link [here](https://github.com/carla-simulator/carla/blob/master/Util/Docker/Release.Dockerfile).

There are a few unofficial ones: 

* [AirSim](https://github.com/abhay-agarwal/airsim-docker-quickstart)

There's also an excellent docker file repo for UE (4.19+), [here](https://github.com/adamrehn/ue4-docker).

I plan on using this to setup a docker file for AirSim. Would a pull request be acceptable?

JihunPark, 2019-03-19:
I agree. It was quite easy to run Airsim on the host machine, but I failed to run Airsim on a docker container.. Please consider providing an official docker file.

madratman, 2019-04-03:
Added support for binaries here https://github.com/Microsoft/AirSim/pull/1868 
I am looking in to building from source as well.

closed 2019-04-03 by madratman