microsoft/AirSim #1599: Multiple SITL in AirSim?

yashkaushal, 2018-12-03:
I've got an SITL instance working in AirSim, How do I run multiple PX4 SITL instances in AirSim?

Thanks

jasonhbartlett, 2018-12-03:
Would be good to know about your existing setup.  A.) Would make it easier for folks to assist with your question.  B.) Would help others who are also trying to get SITL working (such as myself :).  

Are you running on Windows or Linux?  Unreal or Unity?  Versions?  What does your settings file currently look like?

yashkaushal, 2018-12-05:
Hi Jason,

In my current setup I'm using two systems, one with windows 10 running PX4 on bash for Windows(ubuntu 16.04) and a separate desktop running ubuntu 16.04. I'm running AirSim 1.2.0 in Unreal 4.20 as multiple SimpleFlight/HITL/SITL is not yet supported in Unity.

Below is my settings file:
```
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "001": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false
    },
    "002": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "LocalHostIp": "192.168.43.66",
      "SitlIp": "192.168.43.156",
      "UdpIp": "192.168.43.156",
      "QgcHostIp": "192.168.43.156",
      "LogViewerHostIp": "192.168.43.156",
      "X": -4,
      "Y": 0,
      "Z": 0
    }
  }
}
```

I have observed that even if the PX4 firmware is not running on the system, AirSim logs that it is connected(same for when it is connected).

Using the above setup I'm able to connect to both PX4 firmwares(visible in terminal), sometimes I'm able to make a drone take off using QGC.

By using QGC for first and MAVROS for second drone, I'm able to make both of them take off but one of them(the one on separate system) gets unbalanced and moves like crazy (in Auto.takeoff mode), Also note that this multi-drone setup only works about half the times and I've not been able to recognise its pattern yet. You'll have to give them both a separate MAV_SYS_ID parameter to make them visible in QGC otherwise they'll sort of merge and flicker between both of their states.

I have tried multiple drones with HITL and it worked, I used 5 PixHawks with PX4. So i guess AirSim should handle multiple SITL as well but it is showing above mentioned unexpected behaviour.

mjisted, 2019-07-24:
Hi Yash
> 
> I have tried multiple drones with HITL and it worked, I used 5 PixHawks with PX4. So i guess AirSim should handle multiple SITL as well but it is showing above mentioned unexpected behaviour.

It would be great to see the json setup file for the multi-vehicle HITL simulation please!

We can spawn several vehicles, but AirSim will only accept the connection for one if connected via serial/USB.

Thanks!

yashkaushal, 2019-08-01:
Hi Mike,
When connecting single flight controller for HITL simulation, AirSim detects the `COM port` and initialises it. In case of multiple devices the user has to provide serial port information along with unique `system ID` of flight controller.
Here is the configuration for reference:
```json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode":"Multirotor",
  "Vehicles": {
    
    "020": {
      "VehicleType": "PX4Multirotor",
      "VehicleSysID": 20,
      "SerialPort":  "COM5",
      "X": 0,
      "Y": 0,
      "Z": -2
    },
    "030": {
      "VehicleType": "PX4Multirotor",
      "VehicleSysID": 30,
      "SerialPort": "COM12",
      "X": 8,
      "Y": 0,
      "Z": -2
    },
    "040": {
      "VehicleType": "PX4Multirotor",
      "VehicleSysID": 40,
      "SerialPort":  "COM8",
      "X": -8,
      "Y": 0,
      "Z": -2
    }
  }
}
```

closed 2020-08-11 by madratman