microsoft/AirSim #1601: vs 2017 build errors

sheepwxy, 2018-12-03:
hello， I use VS2017 to build the AirSim, but it occurs some build errors.

**In the process of building,the wrong message shows as below:**

"D:\software\Unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50):fatal error C1001 : An internal error occurred in the compiler.[D:\software\Unreal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
(Editor file"D:\agent\_work\3\s\src\vctools\compiler\cxxfe\s1\p1\c\types.c" Line4563)

**At the end,the wrong messages shows as below:**

“D:\software\Unreal\AirSim\AirSim.sln”(default project) (1) ->
“D:\software\Unreal\AirSim\DroneShell\DroneShell.vcxproj”(default project) (2) ->
“D:\software\Unreal\AirSim\AirLib\AirLib.vcxproj”(default project) (3) ->
“D:\software\Unreal\AirSim\MavLinkCom\MavLinkCom.vcxproj”(default project) (4) ->
(ClCompile 目标) ->
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\software\unreal\airsim\mavlinkcom\common_utils\type_utils.hpp(50): fatal error C1001: An internal error occurred in the compiler。 [D:\software\Unr
eal\AirSim\MavLinkCom\MavLinkCom.vcxproj]

bmalhigh, 2018-12-03:
I am assuming you upgraded your Visual Studio to 15.9.x.
There is an issue with VS 15.9.x...This pull request added a workaround for it
https://github.com/Microsoft/AirSim/pull/1558 

Did you pull the latest AirSim code?

sheepwxy, 2018-12-04:
> 
> 
> I am assuming you upgraded your Visual Studio to 15.9.x.
> There is an issue with VS 15.9.x...This pull request added a workaround for it
> #1558
> 
> Did you pull the latest AirSim code?

Thanks for your help, I have solved this problem.

decanbay, 2019-03-14:
Thanks, you saved my day

DRAEYE, 2019-05-10:
Excuse me ,can you tell how to solve this problem?

sheepwxy, 2019-05-10:
> 
> 
> Excuse me ,can you tell how to solve this problem?

Are you using VS2017 15.9 Version? I changed the VS Version and solved this problem.