microsoft/AirSim #1602: Ubuntu Px4 hitl is crashing

YuCodes, 2018-12-03:
* Ubuntu 16.04LTS Airsimv1.2.0Linux UnrealEnginev.4.18.3 

I'm trying to get the PX4 running from my pixhawk plugged into USB, i can arm and disarm and do all the things in qgroundcontrol .
But when i'm trying to use HITL, Airsim connects with the serial port and does nothing, it gives an error in the console and simulation gets stuck in the window.
Exception occurred while updating world: Message length 56 doesn't match expected length64

my setting.json file is 
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  
  "Vehicles": {
		"PX4": {
      "VehicleType": "PX4Multirotor",

      "LogViewerHostIp": "127.0.0.1",
      "LogViewerPort": 14388,
      "OffboardCompID": 1,
      "OffboardSysID": 134,
      "QgcHostIp": "127.0.0.1",
      "QgcPort": 14550,
      "SerialBaudRate": 115200,
      "SerialPort": "*",
      "SimCompID": 42,
      "SimSysID": 142,
      "SitlIp": "127.0.0.1",
      "SitlPort": 14556,
      "UdpIp": "127.0.0.1",
      "UdpPort": 14560,
      "UseSerial": true,
      "VehicleCompID": 1,
      "VehicleSysID": 135,
      "Model": "Generic",
      "LocalHostIp": "127.0.0.1"
    }
}
    }

YuCodes, 2018-12-03:
solved it .

zardosht, 2023-10-26:
This is related to the version of MAVLink protocol that is being used (version 1.0 or version 2.0). In the PX4 settings, you need to change the `MAV_PROTO_VER` parameter in PX4 to corresponding version. Setting it to `2: Always use version 2` fixed the problem for me. 
Corresponding line: https://github.com/microsoft/AirSim/blob/6688d27d3712c2a9c824ababec7a2703475b6628/MavLinkCom/src/MavLinkMessageBase.cpp#L58