microsoft/AirSim #1605: How to transform the world coordinates of an object to 2D coordinates on the camera frame?

flyers, 2018-12-04:
Hi,

I would like to get the 2D coordinates of an object in a certain camera frame. So far I know that the simGetObjectPose() can return the 3D pose of the object in the world frame and the simGetCameraInfo() can return the 3D pose of the certain camera, along with a 4 by 4 projection matrix. I am not sure what exactly is this 4 by 4 matrix because the camera projection matrix should be 4 by 3 usually. So how should I combine those things to get what I want.
Does anyone have any ideas on it? Thanks for your help.

flyers, 2018-12-05:
@jasonhbartlett Thanks for your help. However I want to achieve this in Unreal.

ironclownfish, 2020-02-14:
If I remember correctly, there is an extra row in the projection matrix that just has a -1 at the end. However our projection matrix is not quite right unless the camera is pointing straight down. You'd have to use the FOV to compute the projection matrix yourself. In the meantime I'm entering this as a bug.

LIU-Xueming, 2021-11-05:
> Hi,
>
> I would like to get the 2D coordinates of an object in a certain camera frame. So far I know that the simGetObjectPose() can return the 3D pose of the object in the world frame and the [...]

Hi  @flyers ,
Have you solved this problem？I am also confused about this 4X4 matrix .
Thanks！